raspberrypi/documentation #2372: Some spelling Style Guide

RuffaloLavoisier, 2022-01-19:
Should I understand most of the English spelling in the technical document based on the [style guide](https://github.com/raspberrypi/style-guide/blob/master/style-guide.md#spelling)? 
Or does this need to be corrected?
`Colour`, `Behaviour`...

aallan, 2022-01-19:
> Should I understand most of the English spelling in the technical document based on the [style guide](https://github.com/raspberrypi/style-guide/blob/master/style-guide.md#spelling)?
> Or does this need to be corrected?
> `Colour`, `Behaviour`...

English spelling, not American, is correct. So "colour" rather than "color", and "Behaviour" rather than "Behavior". We're a British company after all!

closed 2022-01-19 by aallan

RuffaloLavoisier, 2022-01-19:
@aallan Thank you for your answer.