raspberrypi/documentation #2373: CM4 Spi3 Dma

lovaselek, 2022-01-19:
Hi,
I read the cm4 documentation and searched all the available sources, but I couldn't find any information about using SPI3 with dma.

The https://datasheets.raspberrypi.com/bcm2711/bcm2711-peripherals.pdf on page 62 does not list the SPI3 in list of DERQ, but Chapter 9 does not write that spi3 could not be used with dma.

Does spi3 also have a bit in the PACTL_CS register? Or it is really not assigned to Spi3 dma. 
Any explanation would be great.

Thanks

aallan, 2022-01-19:
Ping @dp111

lurch, 2022-01-19:
ping also @pelwell 
Although I suspect the answer is that if SPI3 isn't listed in "4.2.1.3. Peripheral DREQ Signals", then the simple answer is that it isn't supported?

pelwell, 2022-01-19:
The relevant bits of PACTL_CS have been documented, and no, there is no DREQ signal for SPI3 (nor SPI2) - so no DMA.

lovaselek, 2022-01-20:
Thanks for the quick response!
Then I will only be able to use it in interrupt mode.

closed 2022-01-20 by lurch