raspberrypi/documentation #2385: BCM2711: AUX_MU_IER_REG bit positions wrong?

MrMino, 2022-01-28:
Latest BCM2711 Peripherals documentation (Jan 18 2022) states the following about `AUX_MU_IER_REG` (when `DLAB=0`) :

<html><head></head><body>

Bit | Name | Description | Type | Reset
-- | -- | -- | -- | --
1 | Enable receive | If this bit is set the interrupt line is asserted whenever the receive FIFO holds at least 1 byte. If this bit is clear no receive interrupts are generated. | RW | 0x0
0 | Enable transmit | If this bit is set the interrupt line is asserted whenever the transmit FIFO is empty. If this bit is clear no transmit interrupts are generated. | RW | 0x0

</body></html>

Poking around, it seems to me like this is not the case - the bit numbers are backwards.

Setting bit 1 in a code that expects an interrupt when a key is pressed in an uart console leaves me with IRQ 125 storming the CPU. Setting bit 0 has the correct behavior - it fires an interrupt only once, when I press the key. The code clears the FIFO and nothing else happens. 

Am I misreading something here or is this a documentation bug?

Kristine1975, 2022-01-29:
This seems to be a documentation bug, dating back to the BCM2835 documentation: https://elinux.org/BCM2835_datasheet_errata#p12

> Bits 1:0 are swaped. bit 0 is receive interrupt and bit 1 is transmit.

You can find a download of the BCM2835 documentation with the errata here: https://github.com/raspberrypi/documentation/issues/325#issuecomment-362877591

lurch, 2022-01-29:
ping @pelwell ?

pelwell, 2022-01-31:
The kernel headers agree: https://git.kernel.org/pub/scm/linux/kernel/git/next/linux-next.git/tree/include/uapi/linux/serial_reg.h#n27
Since interrupts are enabled for ttyS0 and it seems to work, bit 0 must be receive and bit 1 transmit.

aallan, 2022-01-31:
Thanks @pelwell. 

@lurch I'll assign this one to you as backlog for next time we update the BCM2711 Peripherals document.

lurch, 2022-05-08:
This is low priority, but should remain open until the "next time we update the BCM2711 Peripherals document".

pelwell, 2022-05-08:
The Raspberry Pi Imager makes it easy to download an EEPROM update - all you need is a spare SD card.

pelwell, 2022-05-08:
Ignore - wrong thread.

lurch, 2022-07-11:
Still in backlog.