raspberrypi/documentation #2394: Dead links

flype44, 2022-02-02:
From :
https://github.com/raspberrypi/documentation/blob/develop/documentation/asciidoc/computers/raspberry-pi/revision-codes.adoc

1/

Dead link :
https://github.com/raspberrypi/documentation/blob/develop/documentation/asciidoc/computers/raspberry-pi/raspberry-pi.adoc#otp-register-and-bit-definitions

Correct one :
https://github.com/raspberrypi/documentation/blob/develop/documentation/asciidoc/computers/raspberry-pi/otp-bits.adoc#otp-register-and-bit-definitions

2/

Dead link :
https://github.com/raspberrypi/documentation/blob/develop/documentation/asciidoc/computers/raspberry-pi/config_txt.adoc#overclocking-options

Correct one :
https://github.com/raspberrypi/documentation/blob/develop/documentation/asciidoc/computers/config_txt/overclocking.adoc#overclocking-options

aallan, 2022-02-02:
If you go to the documentation site (see https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#raspberry-pi-revision-codes) which is the HTML generated from the Asciidoc source, you'll see that both those links go to the correct places. 

Unfortunately, Github doesn't render Asciidoc links correctly.

closed 2022-02-02 by aallan

aallan, 2022-02-02:
Comes back to this issue, https://github.com/raspberrypi/documentation/issues/2005. Which is that Github doesn't support Asciidoc `include` statements, which is something our documentation makes use of fairly extensively.