raspberrypi/documentation #2398: Replacing omxplayer examples in documentation

aallan, 2022-02-03:
`omxplayer` was removed in Bullseye. Examples [using it](https://www.raspberrypi.com/documentation/computers/os.html#playing-audio-and-video) need to be replaced. See discussions in #2395.

lurch, 2022-02-03:
See also #2314

omarecd, 2022-02-19:
Dear team.
I'm a big Fan of raspberry Pi and following the documentation and I was confused...

If
_The simplest way of playing audio and video on Raspberry Pi is to use the installed OMXPlayer application._
Then
Why was it removed ? :(

I just tried the "vlc example.mp3" and it doesn't work straightforward... Could you remove legacy information and put updated one?

Thanks for the great work.

omarecd, 2022-02-19:
Ps.
I'm trying "mpg123 example.mp3" and it works like a charm :)
That could be also an option

lurch, 2022-02-21:
omxplayer uses the 'legacy' graphics stack, which isn't supported on 64-bit. See the section about KMS in https://www.raspberrypi.com/news/raspberry-pi-os-debian-bullseye/

omarecd, 2022-04-24:
Dear Raspberry Pi Team.
You're doing a great job, it's just that as the situation is still the same (examples need to be updated) I am not sure if closing the issue would be correct. I say this because I received a mail telling me that this would be closed.
Thanks for the great support.

aallan, 2022-04-24:
Yes, still needs to be done. Unfortunately, we're a small team, and it hasn't reached the top of the pile. We welcome pull requests!

lurch, 2022-07-02:
Still needs to be done.

JamesH65, 2022-09-04:
ping

aallan, 2023-04-27:
Part of the hanging chad from the buster to bullseye move that needs to be done for bookworm.

aallan, 2023-07-02:
Still a thing!

lurch, 2023-07-06:
In progress; I'm in the middle of documenting some of VLC's command-line option equivalents to omxplayer.

lurch, 2023-08-21:
Fixed on the `bookworm` branch. OMXPlayer is still mentioned in `documentation/asciidoc/accessories/tv-hat.adoc`, but that'll get fixed as part of #2936

commit 98ef314e (2023-08-21): Update omxplayer instructions to VLC instead.
Fixes #2398
closed 2023-08-21 by lurch