raspberrypi/documentation #2405: About style-guide

RuffaloLavoisier, 2022-02-05:
It's a small question according to the document correction.
According to the [style guide](https://github.com/raspberrypi/style-guide/blob/master/style-guide.md#raspberry-pi-computers-microcontroller-devices-and-microcontroller-boards), I understood that it should be corrected according to the items below. 
Is that right?

1. `Pi 3B`, `Pi 4` -> `Raspberry Pi 3B`, `Raspberry Pi 4`
2. `the Raspberry Pi 4` -> `Raspberry Pi 4`
3. `Raspberry Pis` (no apostrophe) ->`Raspberry Pi computers`, `Raspberry Pi devices` (better)

aallan, 2022-02-07:
> 1. `Pi 3B`, `Pi 4` -> `Raspberry Pi 3B`, `Raspberry Pi 4`

Unless there's a decent reason — where the difference is *important* — I wouldn't differentiate between Raspberry Pi 3, Model B, and Raspberry Pi 3, Model A. I'd just refer to them both as "Raspberry Pi 3."

> 2. `the Raspberry Pi 4` -> `Raspberry Pi 4`

Yup. We hate prepositions here at Raspberry Pi. I don't know why. 🤷‍♂️ 

> 3. `Raspberry Pis` (no apostrophe) ->`Raspberry Pi computers`, `Raspberry Pi devices` (better)

I usually use "Raspberry Pi boards" or "Raspberry Pi computers" depending on context. Now we have a microcontroller class device, RP2040, the word "device" can be confusing as it has a specific technical meaning when we're discussing those.

RuffaloLavoisier, 2022-02-08:
@aallan 
Thank you, in addition to this case, is it necessary to change it? 
ex) `3B`, `4B`, `rpi` -> *Which word would be appropriate?*

aallan, 2022-02-08:
Really depends on the context.

RuffaloLavoisier, 2022-02-08:
@aallan 
Tnx, I will revise it according to the issue.

closed 2022-02-08 by aallan