raspberrypi/documentation #2409: Improve punctuation for channel_config_set_dreq() doc

blippy, 2022-02-07:
The description for channel_config_set_dreq() is confusing to read in the PDF. 

Instead of
> Sources for TREQ signals are internal
(TIMERS) or external (DREQ, a Data Request from the system). 0x0 to 0x3a → select DREQ n as TREQ 0x3b → Select
Timer 0 as TREQ 0x3c → Select Timer 1 as TREQ 0x3d → Select Timer 2 as TREQ (Optional) 0x3e → Select Timer 3 as
TREQ (Optional) 0x3f → Permanent request, for unpaced transfers.

use a better layout, e.g.

> Sources for TREQ signals are internal
(TIMERS) or external (DREQ, a Data Request from the system). 0x0 to 0x3a → select DREQ n as TREQ  
**,**
> 0x3b → Select Timer 0 as TREQ
**,**
> 0x3c → Select Timer 1 as TREQ,
**,**

etc. (The limitation of github prevents me from making a proper formatting suggestions.)

aallan, 2022-02-07:
Sorry, but which PDF document is this?

blippy, 2022-02-07:
Pico C SDK. Section 4.1.7.2.4 of v1.7.1. 

Comma-separation between the values (or putting it in a table) helps to delineate each case of the timer. Otherwise the sentence flows continuously and it's difficult to tell which value refers to which timer.