raspberrypi/documentation #2425: Steps for compiling a 64-bit kernel doesn't mention arm_64bit=1 option requirement

xRiot, 2022-02-13:
I've been struggling for a few days with getting my cross-compiled v5.15 ARM64 kernel to boot on my Raspberry Pi 4 Model B (https://github.com/MichaIng/DietPi/issues/5268). The documentation didn't mention that `arm_64bit=1` should be specified in `/boot/config.txt` :see_no_evil: :
https://www.raspberrypi.com/documentation/computers/linux_kernel.html

lurch, 2022-02-14:
Not something I know anything about, so pinging @pelwell for advice.

pelwell, 2022-02-14:
Is it unreasonable to assume that somebody should know how to boot a 64-bit kernel before they attempt to run one that they have built? Should it also mention how to get to a command shell?

It isn't strictly necessary to use `arm_64bit=1` - only if the equivalent 32-bit kernel is also present.

closed 2022-02-14 by aallan

xRiot, 2022-02-14:
> Is it unreasonable to assume that somebody should know how to boot a 64-bit kernel before they attempt to run one that they have built? Should it also mention how to get to a command shell?
> 
> It isn't strictly necessary to use `arm_64bit=1` - only if the equivalent 32-bit kernel is also present.

You're right, the instructions aren't supposed to teach the reader how Linux works. However, I've personally got 10 years of experience with Linux systems and RPi is my first hands on experience with a ARM64 SBC. Yet I didn't know that you need to explicitly specify a option outside of the kernel compilation to get the system to even boot, it isn't common knowledge.

The documentation currently reminds the user to copy the kernel image to `/boot` partition and the fact that `nproc` can be used to speed up `make` - isn't it reasonable to mention this critical option (maybe at least as a note)? **The fact that system does not boot or produce any journalctl/dmesg log entries** if `arm_64bit=1` isn't specified in `/boot/config.txt` should be good enough reason to add it to the documentation: https://www.raspberrypi.com/documentation/computers/linux_kernel.html

I only posted this issue in hope of saving someone else from wasting time on debugging such issue as everything works perfectly good now on my end :slightly_smiling_face: 

@aallan @pelwell

lurch, 2022-02-14:
Probably also worth noting that the documentation in this repo is targeted primarily at users of Raspberry Pi OS, so if things don't work on some-other-OS that you're using, that's probably a problem for the documentation of that other-OS? :shrug:

aallan, 2022-02-14:
> Probably also worth noting that the documentation in this repo is targeted primarily at users of Raspberry Pi OS, so if things don't work on some-other-OS that you're using, that's probably a problem for the documentation of that other-OS? 🤷

This is my main objection to adding this to the documentation. It's intended to support users running Raspberry Pi OS.