raspberrypi/documentation #2454: Add documentation for non-interactive use of raspi-config

aallan, 2022-03-02:
Need to document non-interactive use of `raspi-config` through using the `nonint` argument. See RPi-Distro/raspi-config#123 for background details.

andrum99, 2022-05-16:
I don't think there should actually be any documentation for `raspi-config` in the documentation repo: the reason it exists is that previously there was no GUI-based config tool on the Pi, and it was deemed useful to document the text-based `raspi-config`. Non-technical users will generally use the GUI-based tool, and for the rest the tool should be obvious enough. Each menu option is effectively self-documented, so I don't see any need for there to be documentation for `raspi-config` in this repo.

andrum99, 2022-05-16:
> > I don't think there should actually be any documentation for `raspi-config` in the documentation repo
> 
> Okay. But as the head of documentation, I _do_ think there should be documentation for `raspi-config` in the documentation repo? 🤔

I'm obviously completely fine with that, and respect that it is entirely up to you. I was just adding my tuppence worth: apologies if that was inappropriate.

aallan, 2022-07-18:
Still on the list.

aallan, 2023-03-02:
The #2766 pull request addresses this issue. I think it needs reworked before merge however.

aallan, 2023-03-02:
Alternative PR #2827 which should close this issue.

PR #2827 (2023-03-16): Documentation for raspi-config command line interface
This PR should close #2454.
closed 2023-03-16 by aallan