raspberrypi/documentation #2457: Search window when testing

RuffaloLavoisier, 2022-03-02:
https://github.com/raspberrypi/documentation/pull/2333

If you build a localhost and go to the desired keyword through the search window, the results window will guide you to the deploy site, not the prototype. 

When testing, In my humble opinion, i think this part needs to be modified to go to the prototype site rather than the deploy site.

lurch, 2022-03-02:
That's not possible, as it's a [cloud-based search system](https://www.algolia.com/doc/guides/getting-started/how-algolia-works/). And the cloud-based system isn't going to be indexing files on your local computer :wink:

closed 2022-03-02 by aallan

RuffaloLavoisier, 2022-03-02:
@lurch 
Thank you for answer 

If it can be implemented in a way other than a cloud-based method in the future, can it be reflected depending on the situation?

aallan, 2022-03-02:
Since we more-or-less just moved from local to cloud-based, see #2333 fro details, I doubt we're heading back the other way any time soon.

lurch, 2022-03-03:
> can it be reflected depending on the situation?

I think you're asking if it can use the Algolia (online) search if building for the online site, or instead use the previous Jekyll (offline) search if building locally for development purposes (e.g. by setting some flag before building the site)? Sure, I guess that's _theoretically_ possible, but it's such a niche use-case we won't be putting in any effort to support that ourselves.

RuffaloLavoisier, 2022-03-03:
@lurch 

https://github.com/raspberrypi/documentation/pull/2333#issuecomment-1009926053

>The only slight drawback (which I guess is unavoidable with any cloud-based search system) is that when I'm viewing http://127.0.0.1:4000/documentation/ for local testing, the search results take me directly to a page on https://www.raspberrypi.com/documentation/ (so we'll just have to remember to double-check which version we're looking at when doing local testing).

Oh, you already mentioned it.