raspberrypi/documentation #2484: Need to document net-boot config options

lurch, 2022-03-16:
Now that net-boot has moved to `stable` https://github.com/raspberrypi/rpi-eeprom/pull/420 we probably ought to document the new config-options related to this.
There's a link in https://github.com/raspberrypi/rpi-eeprom/pull/415 to a PDF which documents the new options.

@peterharperuk / @timg236 Is that [PDF](https://drive.google.com/file/d/1Ym2gfdusW5rs-LVawu5EBrdYs_EZY9if/view) up-to-date with all the necessary options?
(Hmmm, I guess it can't be, because it doesn't mention the `IMAGER_REPO_URL` option mentioned [here](https://github.com/raspberrypi/rpi-eeprom/pull/415#issuecomment-1055594644) ? )

peterharperuk, 2022-03-16:
Hi - I'm litterally working on a commit for the documentation as we speak :)

aallan, 2022-03-16:
Thanks @peterharperuk

closed 2022-03-17 by aallan