raspberrypi/documentation #2487: rfkill will soft block wifi on headless setup unless /boot/wpa_supplicant.conf is present.

esanchezvela, 2022-03-20:
I've setup several images of raspbian, being bullseye the latest, and I have found that unless the /boot/wpa_supplicant.conf file is present, rfkill will soft block the wifi adapter.

All is fine, if that is what expected, however I would like to have seen in the documentation instead of buried in some random community site.

spl237, 2022-03-21:
Although I do know that wifi is soft-blocked until a country code is set, and that is in wpa_supplicant.conf, so it sounds right to me...

XECDesign, 2022-03-21:
It will be blocked until the user tries to connect to an access point using any of the following methods:

1. Dropping a /boot/wpa_supplicant.conf file (intended for headless setups)
2. Raspberry Pi Imager's advanced setting menu
3. raspi-config
4. The desktop applet

If you don't do any of those, you'll see a nag message every time an interactive session runs telling you what you need to do, so that's not really "buried".

aallan, 2022-03-21:
Looking at https://www.raspberrypi.com/documentation/computers/configuration.html#configuring-networking-2 it seems to already be fairly explicitly called out in the documentation?

closed 2022-03-21 by aallan

esanchezvela, 2022-03-21:
my bad, It indeed does, would you consider updating the code sample to make it more obvious?