raspberrypi/documentation #2490: dtoverlay uart pins

pbevilac, 2022-03-21:
[Docs here](https://www.raspberrypi.com/documentation/computers/configuration.html#cm1-cm3-cm3-and-cm4) says: 
_You must also specify which GPIO pins to use, for example_

That is not correct. Specifying GPIO pins is optional, To find what are the default pins used, enter dtoverlay -help uart0, replacing 0 with the UART number of your interest. In addition, UARTs above 0 and 1 do not allow changing GPIO pins.

aallan, 2022-03-21:
Ping @dp111, this is wrt UART on CM1, CM3, CM3+ and CM4 I think?

pbevilac, 2022-03-21:
Just to clarify. The section I quoted above is under the header **CM1, CM3, CM3+ and CM4**

But in fact, specifying pins is not a must for UART 0 and 1, and, one can't change them for UARTs numbered above 1.

aallan, 2022-03-21:
Fixed in 021fe13

closed 2022-03-21 by aallan