raspberrypi/documentation #2494: No Documentation on how to set up SSH Key in imager

RFBomb, 2022-03-23:


lurch, 2022-03-23:
ping @maxnet for clarification of where the auto-filled-in SSH key value comes from.

maxnet, 2022-03-23:
>The field came pre-filled with a single-line key.

Then I am guessing your operating system is either Linux or Mac OS X.
And you did run the "ssh-keygen" command at some point before.
We use the contents of ~/.ssh/id_rsa.pub to prefill the field, that is generated by that.

That file would be used if you open a terminal and type on the command-line a line like "ssh pi@raspberrypi.local".
So you may want to try if that does work.

It is not necessarily the key alternative SSH clients like Putty use. So you may need to replace it if you are using that.

>I tried setting up a new image of Octoprint

Octoprint is a software package offered by multiple operating systems.

If you mean that you installed the OctoPi OS that bundles Octoprint, you may want to check with @guysoft if that implemented support for OS customization for public keys.
I think he was cherry picking values from our firstrun.sh in his Linux distribution.

aallan, 2022-03-23:
So after talking with some internal folks the way this works is that if you are working on a Linux (or macOS) computer, and generate a user SSH key by running `ssh-keygen` then the OS will generate a key pair, and stick the private key in your home directory in `~/.ssh/id_rsa` and a public key in `~/.ssh/id_rsa.pub`

When you run Imager on Linux, or macOS, it will take the contents of `~/.ssh/id_rsa.pub `and prefill the SSH public key field with it. So copying and pasting is not necessary, just checking the "enable SSH" checkbox should be enough to activate it.

If the user then `ssh pi@raspberrypi.local`, it should then login automatically without any password prompt. Which is more or less how I thought it worked, except I figured that you'd have to copy&paste into the SSH key field and imager, and apparently that isn't necessary. Which is nice.

The guess as to why you're having problems @RFBomb is that — if you're not running Octoprint on Raspberry Pi OS, and installing a custom OS image, not the official Raspberry Pi image — is that image just doesn't support being configured in this way. Not all operating systems do.

The alternative explanation is that it's something to do with the fact you're using the command-line ssh client, but an alternative ssh client and things just aren't working for that reason.

I'll add a PR to put some of this into the documentation. But you should go ahead and burn an official Raspberry Pi OS image and see if things "just work" if you do that!

RFBomb, 2022-03-23:
> Then I am guessing your operating system is either Linux or Mac OS X.

I'm running windows 10 64-bit. 

I tried with Putty to no avail. And then I opened up Command Prompt and used 'ssh pi@192.168.1.105' and it didn't work. Maybe I forgot the `pi@`, but I don't think I did. I use putty for all my SSH sessions, so I was surprised it didn't 'just work' when using this. Even when told it to not use the previously working custom key I generated. 

As for my build, I was flashing the base Octoprint image the tool provides. Their docs say this all works (well, say you can set this setting if you want to, no mention of it not working would be more accurate to say). 

I'll try it again tonight. 

As far as using a custom key, do I simply paste the generated public key into that field?

maxnet, 2022-03-23:
>I'm running windows 10 64-bit.

Strange. Not sure where the pre-filled key comes from then. The file location we are looking at normally wouldn't exist on Windows.
Are you sure you didn't set it yourself earlier, and chose to save settings persistently then?
Or did you use software like mingw/msys/cygwin at some point?

>As far as using a custom key, do I simply paste the generated public key into that field?

Yes.
Should be a line that looks like "ssh-rsa somelongbase64string"

Suggest you try using it with an RPI OS image first.

RFBomb, 2022-03-23:
I used the built in key generator tool. 

ssh-keygen has existed on windows 10 since October 2019

maxnet, 2022-03-23:
>As for my build, I was flashing the base Octoprint image the tool provides.

We offer more than one OS image that bundles Octoprint.

[image]

>ssh-keygen has existed on windows 10 since October 2019

Ah, didn't know that.
They don't have a "ssh" client program that comes along with it (instead of Putty)?

RFBomb, 2022-03-23:
Most of those options are the base 'OctoPi' image with customization. By 'base' I referred to 'Octopi', sorry for confusion. 

Yes, you can call 'ssh pi@address' directly from windows 10 command prompt. I also use it (if putty fights me) to connect to an OpenMediaVault Debian server.

aallan, 2022-03-23:
So my guess is that the OctoPi image you're using just doesn't support this customisation via Imager.

guysoft, 2022-03-23:
Hey, 
I just tested the OctoPi stable image from the imager, I pasterd my ``id_rsa.pub`` key in the the field. And I also changed the hostname.

I could ssh to the newly flashed image using my ssh key.
So it works for me.

OP asks about PPK, which is a different format for public keys.
pub is an SSH public key in OpenSSH format, aka the server that rpi runs. ppk file is Putty's private key which is a diffren format.

The imager just sticks that string in authorized hosts. here is the line code that does that:
https://github.com/raspberrypi/rpi-imager/blob/qml/src/OptionsPopup.qml#L654

aallan, 2022-03-23:
I'll leave this open as a placeholder to remind me to add documentation around this.

RFBomb, 2022-03-24:
Ok, so for what its worth, Prior to re-imaging it, I tried using SSH from Win10 command prompt. 
Last night when I tried it, I believe I may have forgotten the `pi@`, since i was able to get in this time. Though I did have to provide the passphrase for the key before it would accept the log in. 

Though Putty was still unable to log in until I manually added my putty key into the pi's allowed keys. Once I did that, Putty worked fine. 

My confusion stemmed primarily from the key being autofilled with no indiciation as to why or how to use it. This discussion was enlightening as far as where it came from. Thanks all

maxnet, 2022-03-25:
>Though I did have to provide the passphrase for the key before it would accept the log in.

Well, then you probably also provided a passphrase when you created the key with ssh-keygen earlier...

I haven't tried on Windows, but on Linux ssh-keygen will ask you 3 questions.
And the trick for passwordless logins, is just pressing "enter" 3 times, to generate a ssh public key at the default location, with the default "empty for no passphrase" passphrase...

```
$ ssh-keygen
Generating public/private rsa key pair.
Enter file in which to save the key (/home/myusername/.ssh/id_rsa): 
Created directory '/home/myusername/.ssh'.
Enter passphrase (empty for no passphrase): 
Enter same passphrase again: 
Your identification has been saved in /home/myusername/.ssh/id_rsa
Your public key has been saved in /home/myusername/.ssh/id_rsa.pub
The key fingerprint is:
SHA256:HunX6AfK2S4rBL526bRmt6xTVxFRoRbzX0Qc1Mmen4s myusername@raspberrypi
The key's randomart image is:
+---[RSA 3072]----+
|            =+=**|
|            .= +o|
|            o.o o|
|    .    . ..  +.|
|   . .  S  .    +|
|    . .o..oo   ..|
|     o.+o=o.. . .|
|    o.Bo*o. .E . |
|   . =+=+=o.     |
+----[SHA256]-----+
```

RFBomb, 2022-03-25:
Yea, I was fine with the password required there, as once I was in and able to insert the line for my putty key (no password required) it all worked

andrum99, 2022-05-07:
Rather than the user having to go looking for documentation to find out where the key has been pre-filled from, it might be better to just explain it in the UI. Perhaps just add something like 'Pre-filled from /home/pi/.ssh/id_rsa.pub'.

aallan, 2022-07-24:
Still needs doing

Sylvanas-1977, 2022-11-04:
Mine keeps failing while if setup same public key on my synology nas it just works, tries with raspery pi imager, it refuses to work invalid public key or whatever pulls hair out.
give me control of my raspery pi i may as well just throw it in the trash or install something else on it cos its not working anyway

maxnet, 2022-11-04:
>invalid public key or whatever

You will have to be a bit more specific about which error you get exactly.
You do can ping the Pi (so are certain it is not having problems with connecting to wifi instead, if you are using that), and are trying to login with the same username you configured in Imager? (if it is the same username as you are using on your normal computer you can do "ssh thepihostname.local", otherwise you typically will need to use a command among the lines of "ssh pi@thepihostname.local")

Sylvanas-1977, 2022-11-04:
Public key denied

maxnet, 2022-11-04:
And you are using a normal RPI OS image (not something else), and specified the same user name in your ssh command line, as you configured in Imager?

Sylvanas-1977, 2022-11-04:
I give up i figure out how to install something of my own

ryanchen2134, 2023-06-05:
Please address this

aallan, 2023-06-05:
> Please address this

We accept PRs

aallan, 2023-09-25:
@nathan-contino So, after the bridged wireless thing. Maybe this? 🤔

maxnet, 2023-09-26:
Note that I added a convenience button to latest Imager.

[image]

[image]

(Only clickable when there is not already a key in ~/.ssh/id_rsa

And users do need to use the standard OpenSSH client that is already installed under Linux/OS X/recent versions of Windows to connect.
E.g. just open a terminal or Windows command prompt and enter "ssh nameofpi.local".
Instead of using third-party stuff like Putty like some tutorials still seem to recommend).

If public key auth could be made the new norm, that hopefully will cut down people making typos in their passwords (and then complaining Imager is broken, as it surely did not configure ssh/account correctly ;-) )

aallan, 2023-09-26:
> Note that I added a convenience button to latest Imager.

Oh nice! Needs adding to the documentation @nathan-contino

aallan, 2023-11-26:
Still a thing.

nathan-contino, 2024-01-18:
Addressed by https://github.com/raspberrypi/documentation/pull/3439

closed 2024-01-18 by nathan-contino