raspberrypi/documentation #2496: Documentation for "Setting Static IP Addresses" is incorrect & at variance with the system manual for dhcpcd

seamusdemora, 2022-03-25:
IAW the instructions given here, I am submitting this issue for discussion before submitting a pull request. Any feedback is appreciated. 

For reasons I do not understand, users are attracted to the idea of using a static IP address on their Raspberry Pi. If they know how DHCP works, and the roles of the DHCP client (`dhcpcd` for RPi OS) and their DHCP server, this is fine. However, many of them do not, and as the [documentation as currently written](https://www.raspberrypi.com/documentation/computers/configuration.html#static-ip-addresses) is incomplete/incorrect, it thereby abets creation of needless issues. In addition, the documentation is directly at odds with the system manual : `man dhcpcd.conf`. 

My issues with the documentation are as follows: 

   1. Manually creating a static IP on a networked host can not be successful as a solitary and isolated event - as is currently suggested. The DHCP server must also be configured to support this ([REF: RFC 2131](https://datatracker.ietf.org/doc/html/rfc2131)); _**i.e.**_ static settings in `dhcpcd` do not get automatically transferred to the DHCP server. 

   2. The system manual `man dhcpcd.conf` contains the following statement in the `static` section (my emphasis): 

   >For IPv4, you should use the **inform** _ipaddress_ option **_instead of_** setting a static address.   

The documentation ignores the **inform** option, instead suggesting that the fixed IP address designation will work in isolation (and it may work for a while which further confuses things). Yes - placement of the **inform** option in the manual is a bit obscure, but that's beside the point. We know **inform** is a better alternative for many users - e.g. those who do not have admin rights on their router/DHCP server. And even use of the `inform` option must carry a warning to the user: choice of the fixed IP addr **must** be coordinated with the DHCP server, otherwise, the user may run afoul of an existing IP address assignment. 

I feel the documentation should be informative wrt the available options, but explain the user's responsibilities and consequences of a bodged configuration. DHCP being a rather involved protocol, it seems better to avoid instructions that lead to confusion, keep things simple, and point out that even with a DHCP address, the system's ZeroConfig Networking stack supports a totally unique network/host name.

lurch, 2022-03-25:
Is this just splitting hairs (i.e. "_theoretically_ you ought to be doing it this way..."), or are the current instructions actually causing problems for people?

ping also @XECDesign

aallan, 2022-03-25:
I think this is making assumptions about the types of LAN than users of the documentation will encounter that aren't perhaps warranted, and the level of user that is reading it. We should avoid preaching a sermon.

XECDesign, 2022-03-25:
The full quote is regarding an example configuration they provide:

> You could also use the `inform6` command here if you wished to obtain more information via DHCPv6.  For IPv4, you should use the `inform ipaddress` option instead of setting a static address.

I may be wrong, since there's ambiguity, but in the context of the previous sentence it sounds like you "If you want to do A, do this. If you want to do A, under condition B, do another thing".

What would happen if you use `inform` and there's no DHCP server at all?

IMHO, trying to teach people about various ways to have a static IP address with various network configurations is a bit outside the scope of the documentation. 99% of users shouldn't be setting a static IP at all and those who are should already know why they're doing it. We don't have access to that information.

But, none of this is up to me, it's just a few thoughts.

seamusdemora, 2022-03-26:
> I think this is making assumptions about the types of LAN that users of the documentation will encounter that aren't perhaps warranted, and the level of user that is reading it. We should avoid preaching a sermon.

@aallan : The "sermon" is not mine - it's what the Internet Standard (RFC) says. It is also what `man dhcpcd.conf` says. The documentation as currently written is misleading. I am not trying to preach, I am not trying to turn the documentation into a networking tutorial. I am trying to make it clear that simply setting a static IP as it's currently documented will not work without additional steps being taken. There are many ways to communicate that to users.

seamusdemora, 2022-03-26:
@lurch 

> Is this just splitting hairs (i.e. "_theoretically_ you ought to be doing it this way..."), or are the current instructions actually causing problems for people? 

I don't think it's splitting hairs; it's not really a fine point I'm trying to make.  It's a rather frequently asked question. I don't have detailed statistics, but on RPi SE there have been approximately [500 questions with `static ip` in the title](https://raspberrypi.stackexchange.com/search?page=1&tab=Newest&q=%22static%20ip%22%20inquestion%3adhcpcd%20is%3aquestion) since 2017. For the questions I've read myself, I get a strong sense that  many - after reading whatever guidance they read - believe this is all that needs to be done. After reading some of them, I can understand why they're confused. I only read the "official documentation" on this point the day I submitted this issue. 

But it's up to you guys - it's got your "brand" on it - it's your documentation, and it's your responsibility. If you think it's OK as written - leave it as is. But I would encourage you to read the references I cited as that is, in my opinion, the [crux of the matter](https://idioms.thefreedictionary.com/crux+of+the+matter).

seamusdemora, 2022-03-26:
Further to the point of this issue: 

1. wrt the `inform` option: [For at least 2 years](https://forums.raspberrypi.com/viewtopic.php?t=265516#p1613983), `dhcpcd` has had some "errant behavior" in logging when this `inform` option is used in `/dhcpcd.conf`. The problem may be resolved now - but the patch is in version 9.4.0 whereas `dhcpcd ver 8.1.2` is the one currently being shipped with `bullseye`. 

2. The logging issue will have to be resolved through an update to `dhcpcd`; but from 8.1.2 ==> 9.4 sounds like far in the future?? 

3. This "errant behavior" in logging when the `inform` option is used doesn't alter the documentation issue here. It remains true that setting a `static ip` in `dhcpcd` must be accompanied by other administrative changes in the network's DHCP service to make it robust, reliable and iaw the RFC. 

4. This has been discussed numerous times in the RPi forums with various conclusions and recommendations. At least [some of the users in this old forum thread](https://forums.raspberrypi.com/viewtopic.php?t=265516&sid=71d78c5a3554c869197bf0c44f475c90) have recognized the issue with declaring a static ip in isolation. And [as shown in this topic](https://forums.raspberrypi.com/viewtopic.php?t=331869&hilit=dhcpcd&sid=2d099587331a9cd6a9c5aadd7ab131b2&start=25#p1987027) - some get it, and some don't. Which group are you in? 

5. Apparently

aallan, 2022-03-28:
> Which group are you in?

I'm in the group that won't be spoken to in that tone. Please moderate it.

seamusdemora, 2022-03-30:
> I'm in the group that won't be spoken to in that tone. Please moderate it.

There was no offense intended. The remark was not directed at you personally, so I'm confused by your reaction.  I trust you can realize that I did not open this issue with the intent of offending you; I opened the issue only because I feel the documentation can be improved significantly with a minor change. But I will be frank with you: your comment about ["preaching a sermon'](https://github.com/raspberrypi/documentation/issues/2496#issuecomment-1078950438) is most disappointing.

lurch, 2022-03-30:
> This "errant behavior" in logging when the inform option is used doesn't alter the documentation issue here.

Ah, I think I'd missed this bit, and I was confused why you were asking us to add something that you'd admitted doesn't work properly.

> It remains true that setting a static ip in dhcpcd must be accompanied by other administrative changes in the network's DHCP service to make it robust, reliable and iaw the RFC.

Please submit what you consider to be a suitable PR, and we can continue the discussion there.

seamusdemora, 2022-03-30:
@lurch 
> Please submit what you consider to be a suitable PR, and we can continue the discussion there.

OK - I can do that, but since you brought it up, what are your thoughts on how we handle the errant logging behavior? FYI, I didn't get a reply from Mr. Marples, and since the [original Issue](https://github.com/NetworkConfiguration/dhcpcd/issues/47) had been closed, I decided to create a [new issue](https://github.com/NetworkConfiguration/dhcpcd/issues/94). I've still heard nothing, but I can see that he may feel it's not his problem any longer... By that I mean he's committed a fix to his repo, but so far as Raspberry Pi is concerned, it's a release that may be in the distant future.  

I suppose what I'm blethering about is this: I'm reluctant to recommend the `inform` option given that it currently overwhelms one's log files. 

Do you know how this is handled? Since RPi is using an older/non-patched version, is the responsibility for patching in RPi instead of `dhcpcd`?

aallan, 2022-03-30:
> IMHO, trying to teach people about various ways to have a static IP address with various network configurations is a bit outside the scope of the documentation. 99% of users shouldn't be setting a [...]

This from @XECDesign up thread 👆 . 

My own opinion is similar. The majority of users should not be setting a static IP address in the first place, if they want their Raspberry Pi to have the same IP address all the time, then what they should be doing is setting their DHCP server on their router to hand out the same IP address based on the board's MAC address.

Potentially we should just remove the section, or add a `WARNING` to the existing instructions to note that setting a static IP address can be potentially harmful to the network. Which should discourage people who don't know the consequences from doing it.

As I mentioned in the [CONTRIBUTING.md](https://github.com/raspberrypi/documentation/blob/develop/CONTRIBUTING.md#type-of-content) file we are deprecating the more generic documentation around using the Linux operating system. The inner workings of DHCPD isn't necessarily Raspberry Pi specific?

seamusdemora, 2022-03-30:
@aallan , @XECDesign 
> My own opinion is similar. The majority of users should not be setting a static IP address in the first place,

It is physically impossible for me to agree more with that statement. I thought I had made that clear, but I suppose things are never as clear as we imagine them. I would put it this way: 

#### ***IF*** you're going to cover ***HOW*** to create a `static ip` assignment, it should be made clear that there are additional steps that are required, and that those steps are beyond the scope of the documentation. 

But of course the simpler course is to never mention `static ip` at all. 

That's the choice here - as I see it at least - remove the passage on `static ip`, or add a *caveat*. I don't advocate a [long-winded tutorial on DHCP](https://raspberrypi.stackexchange.com/a/136725/83790); there are other places for that.

If we're all in agreement, what is the consensus on how to proceed - in or out?

lurch, 2022-03-31:
> add a WARNING to the existing instructions to note that setting a static IP address can be potentially harmful to the network.

That'd get my vote :+1:

aallan, 2022-03-31:
I've added in some more warnings. See 3b199cb.

closed 2022-03-31 by aallan