raspberrypi/documentation #2505: 3 long 4 short flash code not documented

cleverca22, 2022-04-08:
there is a 3-long + 4-short error code within recovery.bin, its not clear what exactly triggers it, but at least one user has experienced it:

https://forums.raspberrypi.com/viewtopic.php?p=1992039#p1992039

jesperkp, 2022-04-09:
> there is a 3-long + 4-short error code within recovery.bin, its not clear what exactly triggers it, but at least one user has experienced it:
> 
> https://forums.raspberrypi.com/viewtopic.php?p=1992039#p1992039

A connection to the serial debug interface gives this during boot:

SIG pieeprom.sig cbed2d774850b6cc668133218f93080c8ee7528c0d09bf8b4d9aba15c1afbe23 1646065651
Reading EEPROM: 524288
Writing EEPROM
+
Update failed
Erasing EEPROM
FATAL error-code 34

lurch, 2022-04-11:
ping @timg236

aallan, 2022-04-11:
Should be added to https://www.raspberrypi.com/documentation/computers/configuration.html#led-warning-flash-codes if it's a valid code.

timg236, 2022-04-11:
The error code means that the secure-boot configuration is not valid. However, in this case what is happening is that the EEPROM write failed for some unknown reason and a bogus error code was returned.

aallan, 2022-04-11:
Thanks @timg236. I've added the error code to the table, fed672d.

timg236, 2022-04-11:
Thanks @aallan I was trying to think of a better name but I think that's sufficient. There should be more verbose output when the real error occurs and I've pushed an internal PR to fix the bogus error code so that will get picked up in the next release.

lurch, 2022-04-23:
> the next release

See https://github.com/raspberrypi/rpi-eeprom/pull/424