raspberrypi/documentation #2513: Change headless documentation

nmcc1212, 2022-04-20:
change https://www.raspberrypi.com/documentation/computers/configuration.html#setting-up-a-headless-raspberry-pi

to reflect the changes outlined here

https://www.raspberrypi.com/news/raspberry-pi-bullseye-update-april-2022/

mainly that `If you are installing Raspberry Pi OS Lite and intend to run it headless, you will need to create a new user account.` now also applies to the full pi os now as well as lite

aallan, 2022-04-20:
That has been done? e.g.
[image]
and
[image]
in the headless section, and if you're talking about the full installation, see this entire section? 

https://www.raspberrypi.com/documentation/computers/getting-started.html#configuration-on-first-boot

closed 2022-04-20 by aallan

nmcc1212, 2022-04-20:
the message says Raspberry Pi OS Lite but this happens on Raspberry Pi OS aswell

aallan, 2022-04-21:
The message says "since you will not be able to create a new user account on first boot". In the full desktop OS, as is documented, a wizard runs on first boot that allows you to create a new user.