raspberrypi/documentation #2514: CAD-Files missing

berberic2, 2022-04-22:
There are CAD files in STEP-format missing for the raspis.

The schematics are insufficient.
The Zero 2 mechanical drawing pdf p.e. does not even have thickness of board or diameter of mounting-holes.

aallan, 2022-04-22:
Raspberry Pi boards are not open source hardware, we normally relase on reduced schematics and mechanical drawings.

closed 2022-04-22 by aallan

berberic2, 2022-04-22:
I don’t consider this a valid point.
Evene for proprietary, fully closed hardware components one can get CADs, simply because they are necessary to integrate a component into a construction.

lurch, 2022-04-23:
It seems _very_ likely that the mounting-holes on the Raspberry Pi Zero 2 W are the same as for the Raspberry Pi Zero and Raspberry Pi Zero W? And the mechanical-drawings for those _do_ mention the mounting-hole diameters.

berberic2, 2022-04-24:
Probably, but that still doesn't replace CAD, so all the heights are still missing. So anyone who wants to integrate it into a CAD has to build a CAD dummy himself, with a caliper, or get one from _somewhere_.

lurch, 2022-04-24:
If you're building products around Raspberry Pi hardware, that's what the Compute Modules are designed for; and they _do_ have the CAD models available that you're asking for (along with detailed datasheets) https://www.raspberrypi.com/documentation/computers/compute-module.html#datasheets-and-schematics

berberic2, 2022-04-24:
Thanks. But:
ⓐ There are people that use raspis for private projects. These people typically use the “normal” raspis, not the compute modules. There are no standard connectors on the compute-module.
ⓑ Even in the company I work, we use  “normal” raspis for all sort of prototypes and not the compute-modules. 
ⓒ As far as I can see, these is a KiCAD-project and not the kind of CAD-Data that is needed. A simple .step-file with the geometry would be sufficient.

aallan, 2022-04-24:
We do offer some STEP files, for CM4 and Pico. See https://datasheets.raspberrypi.com/. AFAIK all mechanical draws and reduced schematics that we offer are listed there.