raspberrypi/documentation #2515: Default pi user has been removed

Steccah, 2022-04-23:


lurch, 2022-04-23:
Yes, see https://www.raspberrypi.com/news/raspberry-pi-bullseye-update-april-2022/ and https://www.raspberrypi.com/documentation/computers/getting-started.html#configuration-on-first-boot

Steccah, 2022-04-23:
Sorry my fault, i wanted to reference this: https://www.raspberrypi.com/documentation/computers/using_linux.html#linux-users

RuffaloLavoisier, 2022-04-23:
@lurch 
Should these changes be reflected throughout the document?

> Yes, see https://www.raspberrypi.com/news/raspberry-pi-bullseye-update-april-2022/ and https://www.raspberrypi.com/documentation/computers/getting-started.html#configuration-on-first-boot

lurch, 2022-04-23:
Ah, right...

RuffaloLavoisier, 2022-04-23:
@lurch 
I want to help you with this part. What should I do?

lurch, 2022-04-23:
Myself and @aallan will be discussing this on Monday (we're both going to be in Cambridge!).

aallan, 2022-04-23:
We may just entirely remove the "Using Linux" section. We're not really in the business of teaching folks to use UNIX, and honestly, the section doesn't do a great job in any case. Perhaps it's had its day and should just be deleted. 🤔

RuffaloLavoisier, 2022-04-23:
@aallan 
I agree with you.  website is full of more extensive material.

aallan, 2022-04-24:
Submitted PR to remove the 'Using Linux' section entirely. See #2519.

closed 2022-04-24 by aallan