raspberrypi/documentation #2520: Document wireless AP limit

aallan, 2022-04-24:
Right now the maximum user/client limit for the wireless firmware isn't in the documentation. Need to add that. Possibly also document how to swap out the firmware to an older version to increase the limit (or at least point at https://github.com/RPi-Distro/firmware-nonfree/tree/bullseye/debian/config/brcm80211/cypress where it's discussed).

lurch, 2022-04-24:
That link also has both `cyfmac43430-sdio.bin` and `cyfmac43455-sdio.bin`, with the 'minimal' version only available for `cyfmac43455`, so I guess the documentation might need to also clarify which models of Raspberry Pi use which wireless module / firmware?

aallan, 2022-07-24:
Still needs doing

beta-tester, 2022-11-07:
as reference there was a thread in another repository about the limit:
[Opinions about Pi 4B/3B+ WiFi features #2853](https://github.com/iiab/iiab/issues/2853])

aallan, 2023-11-26:
Still a thing.

aallan, 2024-04-03:
The maximum number of clients with the default firmware seems to be between 7 and 10. There is an alternative "minimal" firmware at https://github.com/RPi-Distro/firmware-nonfree/tree/bullseye/debian/config/brcm80211/cypress which supports 19 clients.

lurch, 2024-04-03:
> There is an alternative "minimal" firmware at https://github.com/RPi-Distro/firmware-nonfree/tree/bullseye/debian/config/brcm80211/cypress

Looks like there might be slightly more up-to-date files in https://github.com/RPi-Distro/firmware-nonfree/tree/bookworm/debian/config/brcm80211/cypress ?

nllho, 2024-11-25:
Hi all, I've read through as much of the context as possible and chatted through this with @nathan-contino, and we've decided to mark this docs request as Won't Do. Summarising the reasoning here:

- The maximum number of clients varies depending on how much RAM is being used by a particular Pi and its firmware ([as explained here](https://github.com/raspberrypi/linux/issues/3010#issuecomment-748484728)), so there is no fixed or reliable number to add to the docs.
- The alternative firmware mentioned in this issue is a workaround provided by a Raspberry Pi engineer to increase the number of supported clients.
- In general, we'll avoid adding workarounds to documentation for the sake of maintainability. Workarounds remain discoverable in the forums and GitHub discussions. If there comes a time when the alternative firmware is officially supported, that would be the time right time to update the docs :)

closed 2024-11-25 by nllho

aleb, 2025-10-05:
Mabye you could mention the possible range for the maximum number number of clients.