raspberrypi/documentation #2537: Update references to the default user/pass

aallan, 2022-05-11:
Now the default user has been removed we need to update all references to a default user/pass,

```
$ git grep -E "\`(pi|raspberry)\`" "*.adoc" | grep -Ev "default (user|pass)"
documentation/asciidoc/computers/configuration/access-point-routed.adoc:If
SSH is enabled on the Raspberry Pi, it should be possible to connect
to it from your wireless client as follows, assuming the `pi` account
is present: `ssh [pi@192.168.4.1](mailto:pi@192.168.4.1)` or `ssh [pi@gw.wlan](mailto:pi@gw.wlan)`
documentation/asciidoc/computers/configuration/raspi-config.adoc:The
`sudo` is required because you will be changing files that you do not
own as the `pi` user.
documentation/asciidoc/computers/configuration/raspi-config.adoc:From
this submenu you can select whether to boot to console or desktop and
whether you need to log in or not. If you select automatic login, you
will be logged in as the `pi` user.
documentation/asciidoc/computers/configuration/securing-the-raspberry-pi.adoc:Once
you have confirmed that the new account is working, you can delete the
`pi` user. In order to do this, you'll need to first change the
autologin user to your new user `alice`, with the following:
documentation/asciidoc/computers/configuration/securing-the-raspberry-pi.adoc:Please
note that with the current Raspberry Pi OS distribution, there are
some aspects that require the `pi` user to be present. If you are
unsure whether you will be affected by this, then leave the `pi` user
in place. Work is being done to reduce the dependency on the `pi`
user.
documentation/asciidoc/computers/configuration/securing-the-raspberry-pi.adoc:To
delete the `pi` user, type the following:
documentation/asciidoc/computers/configuration/securing-the-raspberry-pi.adoc:This
command will delete the `pi` user but will leave the `/home/pi`
folder. If necessary, you can use the command below to remove the home
folder for the `pi` user at the same time. Note the data in this
folder will be permanently deleted, so make sure any required data is
stored elsewhere.
documentation/asciidoc/computers/configuration/securing-the-raspberry-pi.adoc:This
command will result in a warning that the group `pi` has no more
members. The `deluser` command removes both the `pi` user and the `pi`
group though, so the warning can be safely ignored.
documentation/asciidoc/computers/configuration/securing-the-raspberry-pi.adoc:and
change the `pi` entry (or whichever usernames have superuser rights)
to:
documentation/asciidoc/computers/getting-started/configuring.adoc:NOTE:
Some older software may require the presence of the `pi` user.
documentation/asciidoc/computers/os/using-gpio.adoc:In order to use
the GPIO ports your user must be a member of the `gpio` group. The
`pi` user is a member by default, other users need to be added
manually.
documentation/asciidoc/computers/os/using-webcams.adoc:If you are not
using the default `pi` user account, you need to add your username to
the `video` group, otherwise you will see 'permission denied' errors.
documentation/asciidoc/computers/remote-access/apache.adoc:This shows
that by default there is one file in `/var/www/html/` called
`index.html` and it is owned by the `root` user (as is the enclosing
folder). In order to edit the file, you need to change its ownership
to your own username. Change the owner of the file (the default `pi`
user is assumed here) using `sudo chown pi: index.html`.
documentation/asciidoc/computers/remote-access/samba.adoc:Firstly,
create a folder to share. This example creates a folder called
`shared` in the `home` folder of the current user, and  assumes the
current user is `pi`.
documentation/asciidoc/computers/remote-access/scp.adoc:Copy the file
`myfile.txt` from your computer to the `pi` user's home folder of your
Raspberry Pi at the IP address `192.168.1.3` with the following
command:
documentation/asciidoc/computers/remote-access/scp.adoc:Copy the
directory `project/` from your computer to the `pi` user's home folder
of your Raspberry Pi at the IP address `192.168.1.3` with the
following command:
documentation/asciidoc/computers/remote-access/vnc.adoc:If you're
connecting from the
https://www.realvnc.com/download/viewer/[compatible VNC Viewer app]
[... 120 lines cut ...]
`main.py`. But you are not limited to Python programs: simply change
the `ExecStart` line to be the command to start any program or script
that you want running from booting.
documentation/asciidoc/computers/using_linux/linux-using-commands.adoc:Using
`cd` changes the current directory to the one specified. You can use
relative (i.e. `cd directoryA`) or absolute (i.e. `cd
/home/pi/directoryA`) paths.
documentation/asciidoc/computers/using_linux/linux-using-commands.adoc:The
`pwd` command displays the name of the present working directory: on a
Raspberry Pi, entering `pwd` will output something like `/home/pi`.
documentation/asciidoc/computers/using_linux/using-the-terminal.adoc:In
order to navigate to other directories the change directory command,
`cd`, can be used. You can specify the directory that you want to go
to by either the 'absolute' or the 'relative' path. So if you wanted
to navigate to the `python_games` directory, you could either do `cd
/home/pi/python_games` or just `cd python_games` (if you are currently
in `/home/pi`). There are some special cases that may be useful: `~`
acts as an alias for your home directory, so `~/python_games` is the
same as `/home/pi/python_games`; `.` and `..` are aliases for the
current directory and the parent directory respectively, e.g. if you
were in `/home/pi/python_games`, `cd ..` would take you to `/home/pi`.
```

the easiest way of fixing these is to use `~/` or `$HOME` instead, but that won't work when an absolute path is needed outside
of a login-session e.g. in the crontab file.

lurch, 2022-05-23:
https://github.com/raspberrypi/Raspberry-Pi-OS-64bit/issues/227 is suggesting that we also need additional tweaks to the places where we've already explained the removal of the default username and password.

bandrews, 2022-05-26:
As someone who primarily flashes their images through balenaEtcher and who exclusively sets up their Pi devices as headless installation, I lost quite a bit of time this evening to complications from this change (first from realizing it had occurred, then from discovering nuances of the userconf.txt file) and would like to offer two documentation suggestions.  

- An information box-style notice on https://www.raspberrypi.com/software/operating-systems/ indicating that headless setup steps have changed would be greatly appreciated.  This page primarily serves users who use their own flashing tools and who therefore are most likely to be affected by the change to no longer provision a default 'pi' user account.

I found Simon Long's comment at [14th April 2022, 8:55 pm](https://www.raspberrypi.com/news/raspberry-pi-bullseye-update-april-2022/#comment-1582940) pretty disappointing, as it indicates the Pi foundation believes that burying this information in the release notes was sufficient notice;  I didn't find that to be the case as a casual user who doesn't follow the details of the foundation's release cadence.  

I can understanding wanting to minimize clutter on the main pages, but I do think a notice on the individually downloadable images page would be appropriate for the foreseeable future.  Something simple and straightforward, like

"Beginning with images released in April 2022, the **pi** user is no longer created by default.  For headless installations, an account can be configured with a [userconf.txt file](https://www.raspberrypi.com/documentation/computers/configuration.html#configuring-a-user)

- The instructions for constructing the necessary password file at https://www.raspberrypi.com/documentation/computers/configuration.html#configuring-a-user were mostly helpful, but omitted a pitfall that tripped me up:  If this file is created from a Windows computer and the single line is terminated with a carriage return/linefeed combination, the user account/password doesn't work.  I surmise this is likely because the carriage return from Windows-style line endings is treated as part of the password hash, as I was finally able to make it work on my fourth flashing attempt by omitting the trailing newline.

Adding a notice to the instructions cautioning users not to insert a newline after their account details, or cautioning Windows users about UNIX-style line endings, would be incredibly helpful and would have saved me an hour of troubleshooting.  (Bonus points if you can pass this feedback upstream to the folks who maintain whatever parses userconf.txt to just fix the underlying issue, of course!)

Thanks for listening to both bits of feedback!  I think this change is overall valuable, but these two adjustments would go a long way towards improving the communication surrounding it, I think.

lurch, 2022-05-26:
> The instructions for constructing the necessary password file at https://www.raspberrypi.com/documentation/computers/configuration.html#configuring-a-user were mostly helpful, but omitted a pitfall [...]

@XECDesign / @spl237 is this fixable for the next OS release?

aallan, 2022-05-26:
> As someone who primarily flashes their images through balenaEtcher 

We recommend our own Raspberry Pi Imager, and the documentation is (quite rightly I think) written from that perspective. We don't necessarily support third-party tools.

> * An information box-style notice on https://www.raspberrypi.com/software/operating-systems/ indicating that headless setup steps have changed would be greatly appreciated. 

Instructions for headless installation were updated the day the release was made, https://www.raspberrypi.com/documentation/computers/configuration.html#setting-up-a-headless-raspberry-pi.

> Adding a notice to the instructions cautioning users not to insert a newline after their account details, or cautioning Windows users about UNIX-style line endings, would be incredibly helpful and [...]

I've added a `[WARNING]` box, see ad6558f.

XECDesign, 2022-05-26:
> @XECDesign / @spl237 is this fixable for the next OS release?

Yes, good point, I should've spotted this. The internal repo now has a fix.

aallan, 2022-05-26:
Thanks @XECDesign! Can you let me know when we push this live and I can remove the WARNING from the documentation?

lurch, 2022-05-26:
I guess the WARNING will need to remain in place until the next OS release (i.e. new download on https://www.raspberrypi.com/software/operating-systems/), as this `userconf.txt` gets consumed before the OS has had the chance to do any updates?

XECDesign, 2022-05-26:
> Thanks @XECDesign! Can you let me know when we push this live and I can remove the WARNING from the documentation?

Yes, I've added a reminder for myself.

> I guess the WARNING will need to remain in place until the next OS release (i.e. new download on https://www.raspberrypi.com/software/operating-systems/), as this `userconf.txt` gets consumed before the OS has had the chance to do any updates?

That's right.

bandrews, 2022-05-26:
> > As someone who primarily flashes their images through balenaEtcher
> 
> We recommend our own Raspberry Pi Imager, and the documentation is (quite rightly I think) written from that perspective. We don't necessarily support third-party tools.

I can understand this sentiment, but I'd like to try one more time to advocate for a quick notice about this breaking change on the download page.  If users have landed on https://www.raspberrypi.com/software/operating-systems/, they already are planning not to use the Pi Foundation's imager;  this page is specifically for those who use other tooling.

And speaking from my own experience, I think these folks are likely to have been long-time users of the Raspberry Pi platform who already have a workflow for setting up new devices, so the notice in the headless setup documentation isn't likely to reach them.  The 'ssh' text file process for headless setup has existed since the early days of the platform, so for 5+ years now I've been in a rhythm of downloading the latest release, flashing it to an SD card, creating the ssh file from memory, and proceeding with setup from there.  Since this workflow is headless, the only opportunity to communicate the breaking change is on the download page.

If the goal is to encourage adoption of the new imager, I think explaining why the foundation would prefer users select it and giving users a reason to adopt it on the download page would be helpful.  Imaging tools are potentially destructive, so I'm hesitant to switch my workflow to a  new tool that has the capability of wiping an entire storage device;  I'm much more open to trying something new if there's a good reason to do so.

If I can make one more attempt at a proposal, perhaps integrating a notice about the user account changes into the text at the top of the page advertising the imager would be something you'd be willing to consider?

"[Raspberry Pi Imager](https://www.raspberrypi.com/software/) is the quick and easy way to install an operating system to a microSD card ready to use with your Raspberry Pi.  As new versions of the operating system no longer create a default user account, the Imager is now the recommended way to configure a headless Raspberry Pi."

Alternatively, choose from the operating systems below, available to download and install manually."

Thanks again for listening and for the rapid response on the line endings portion of the feedback!

aallan, 2022-05-26:
> Imaging tools are potentially destructive, so I'm hesitant to switch my workflow to a new tool that has the capability of wiping an entire storage device

Which is why we only document our own tool.

bandrews, 2022-05-26:
> > Imaging tools are potentially destructive, so I'm hesitant to switch my workflow to a new tool that has the capability of wiping an entire storage device
> 
> Which is why we only document our own tool.

To be clear, I'm not asking for documentation about how to use alternative imaging tools - just suggesting a notice about a breaking change to the operating system's setup process on the download page would be apppropriate.

If anything, this is an opportunity you could take to encourage people to switch to your tooling by highlighting one of the benefits!

FWIW, I'm not the only one with the feedback that a bit easier to find notice would have been helpful, I think:  the [comments threads on the announcement post](https://www.raspberrypi.com/news/raspberry-pi-bullseye-update-april-2022/#comment-1584000) and [discussions on Reddit](https://www.reddit.com/r/linux/comments/u137jl/raspberry_pi_os_discards_the_default_user_pi/i4cyjkh/) both have examples of others with a similar perpsective.

I know your team has likely been absorbed in this change for some time and it's definitely been messaged through blog posts and the headless documentation, so I can understand how you might feel it's been sufficiently documented.  But [many people have an existing workflow that predates the Imager entirely](https://www.reddit.com/r/linux/comments/u137jl/raspberry_pi_os_discards_the_default_user_pi/i4aec3o/), and I think this is an excellent opportunity to help evangelize the new tooling to longtime Raspberry Pi advocates while helping better communicate a major change to the setup process.

As before, thanks again for listening.

lurch, 2022-06-07:
I can appreciate that this must be frustrating for some of you, but
a) as you can see from the big list of file-names in the first comment, we're aware that there's still some pages that need updating, but we're unfortunately snowed under with other stuff at the moment, so this currently isn't top-priority.
b) IMHO changes to RPi Imager wouldn't be appropriate because most people run their Raspberry Pi connected to a monitor, and when you do that (with either the Desktop or Lite versions of Raspberry Pi OS) you get a helpful wizard prompting you to create a new user and set a password.

lurch, 2022-08-07:
Still a valid issue.

XECDesign, 2022-09-07:
> Thanks @XECDesign! Can you let me know when we push this live and I can remove the WARNING from the documentation?

That's up now

lurch, 2022-09-07:
>> Thanks @XECDesign! Can you let me know when we push this live and I can remove the WARNING from the documentation?

> That's up now

Cheers, removed (now-redundant) warning in #2616

jehlers42, 2023-09-18:
Just for clarification, this covers any and every reference to a default user, default password, or both?

lurch, 2023-09-19:
I think both. Several years ago Raspberry Pi OS defaulted to coming with a `pi` user with the password set to `raspberry` (and this is why they've ended up scattered throughout the documentation).
Things have now changed (see https://www.raspberrypi.com/news/raspberry-pi-bullseye-update-april-2022/ ) so that the user is asked at first boot-up to choose a username and password - and if they choose `pi` or `raspberry` they get warned that this is insecure. IIRC you can also pre-configure the username and password using Raspberry Pi Imager, and this then skips that part of the first-boot-wizard?

jehlers42, 2023-09-20:
> When we get round to tackling this we should also refactor the https://www.raspberrypi.com/documentation/computers/configuration.html#changing-your-username part into "Adding a new user".

I like the idea of an "Adding a user" section, but this is currently under the "Securing your Pi" heading and I'm not sure it fits with a new scope. The only reason I could see keeping it where it is, is if we recommend having an "admin" account, and an "everyday" account (like some folks do on Windows or macOS). I don't think that's really common in Linux, however.

Not sure we have any other sort of home for user management tasks, aside from [this page](https://www.raspberrypi.com/documentation/computers/configuration.html#configuring-a-user) in the Headless section. We could always create a new page in the Configuration chapter for rudimentary user management tasks (add, edit, delete).

Thoughts? @aallan, @lurch, and @nathan-contino

lurch, 2023-09-21:
> We could always create a new page in the Configuration chapter for rudimentary user management tasks (add, edit, delete).

That would be my choice, but we probably only need add and delete, not edit.

aallan, 2023-11-26:
Still a thing.

closed 2023-12-08 by aallan