raspberrypi/documentation #2551: Pinout wrong for newer 3b using gpio boot mode

peterhassing, 2022-05-26:
On the newer raspberry pi 3b with the bmc2837B0 the gpio pins correspond to the 3b+, not the 3b.
This https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#pin-assignments part of the documentation should be changed to reflect this.

lurch, 2022-05-26:
ping @jimbojr @Roger-Thornton @pelwell

lurch, 2022-06-09:
Thanks Gordon :+1:

peterhassing, 2022-06-09:
Well, it seems I jumped to conclusions a bit too quick..

Summary:
I'm using the raspberry pi 3b (NOT the 3b+)
Until now these had the BCM2837R1F8G on it.
Now it seems this last batch has the BCM2837B01FSBG.

After burning the otp bits the pi's would not boot anymore:
program_gpio_bootmode=1
program_usb_boot_mode=1

This is while connecting pin 1 (3v3) and pin 15 (gpio22) for sd-card boot, or while connecting pin 1 (3v3) and pin 37 (gpio 26) for USB boot.

In my frustration ( I burned the otg bits on 20 pi's ) I found that sd-card boot works if I connect pin 1 (3v3) to pin 38 (gpio 20) and therefore (maybe wrongly) assumed the gpio-bootmode pins followed the pi 3b+ layout.
USB booting still does not work, also not when I connect pin 1 (3v3) and pin 22 (gpio25).

lurch, 2022-06-09:
Just to clear up any potential confusion, I assume you're aware that `program_gpio_bootmode=1` isn't **required** in order to boot from USB? I've never experimented with that setting myself, but AFAIK it's only there for if you want to circumvent the usual [boot sequence](https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#boot-sequence).

> USB booting still does not work, also not when I connect pin 1 (3v3) and pin 22 (gpio25).

Sounds like another question for @ghollingworth 
I think if I was in your situation I'd try bringing all of GPIOs 20-26 high, and see if USB booting then worked? (and if it does, then try each GPIO one at a time)

peterhassing, 2022-06-09:
There is no confusion. 
These pi's will be potted in epoxy as they are mounted on vibrating diesel engines. 
Only the usb-side of the pcb sticks out together with a switch to select boot-medium. 
This way I can recover from a config or sd failure.

I've tried all gpio's, it's not booting from usb.

lurch, 2022-06-09:
Re-opening, pending further comments / clarification from Gordon.

JamesH65, 2022-08-14:
I believe this should remain open as there are still outstanding issues in the docs?

lurch, 2022-08-14:
ping @ghollingworth 

@peterhassing just to clarify things in this long-winded thread, could you please clarify exactly what is and isn't working? Thanks.

peterhassing, 2022-08-15:
@lurch 
Well exactly as I wrote in my last post.. https://github.com/raspberrypi/documentation/issues/2551#issuecomment-1150844488 
So usb-boot does not work (pulling ANY gpio high) and sd-boot works if I use the pinout 3b+ layout (pin 38 high)
I would love a solution though, as I still have these pi's waiting to be potted as soon as I can boot them from usb..

ghollingworth, 2022-08-15:
I think the above information is correct as far as the bootrom is concerned.  

I'm not sure why it is not working in this particular situation and don't at the moment have the time to do a complete investigation of the bootrom to identify whether something is going wrong either in the firmware (when programming the OTP) or in the bootrom itself.

You say that the SD card GPIO is working if you use GPIO20 and pull it to 3v3
If you let it go low then it should fall back to ignore all boot modes, can you check the voltage on each of the GPIO20 - 26 to make sure they are all correctly pulled low (the default is to pull these low in hardware).
If everything is zero then that means the bootrom will fall back and go to sleep waiting for a new reset
If you drive GPIO25 high before booting (wiring to 3v3 would do), then it should try to boot over USB host as a mass storage device.  If you drive GPIO26 high then you should see if enable the Ethernet.  I assume you're seeing neither of these occurring?

ghollingworth, 2022-08-15:
Ah so you've set it to bank 2 rather than bank 1?

peterhassing, 2022-08-16:
No, it is set to gpio 20, which is pin 38.

ghollingworth, 2022-08-17:
Ok,

In which case:

If you let it go low then it should fall back to ignore all boot modes, can you check the voltage on each of the GPIO20 - 26 to make sure they are all correctly pulled low (the default is to pull these low in hardware).

If everything is zero then that means the bootrom will fall back and go to sleep waiting for a new reset

If you drive GPIO25 high before booting (wiring to 3v3 would do), then it should try to boot over USB host as a mass storage device. If you drive GPIO26 high then you should see it enable the Ethernet. I assume you're seeing neither of these occurring?

aallan, 2022-10-21:
The PR associated with this issue #2555 got merged 4 months ago. Is this solved now? Is there anything outstanding from this that still needs fixing?

PR #2555 (2022-06-09): Update GPIO boot-mode for Pi3B with BCM2837B0
Fixes #2551
commit d2eb1fb3 (2022-06-09): Update GPIO boot-mode for Pi3B with BCM2837B0

Fixes #2551
closed 2023-01-01 by github-actions[bot]