raspberrypi/documentation #2564: Issue importing images

tobybroberts, 2022-06-24:
make doesn't seem to work properly when adding new images.

lurch, 2022-06-24:
Something isn't getting re-scanned properly, which means if you add a new image and then reference it in an adoc file, the appropriate auto-generated dependencies file doesn't get updated; and thus when you run `make` the new image doesn't get copied from the source directory to the output directory, and thus doesn't appear in the rendered site when you run `make serve_html`.
Until this gets fixed, the workaround for now is to just run `make clean` before running `make`.

JamesH65, 2022-08-28:
ping

lurch, 2022-08-30:
This _will_ need fixing at some point, but as we don't add new images to the documentation very often, it currently isn't top of the todo-list.