raspberrypi/documentation #2565: How can I get the Address distribution of 1 GB memory for BCM2837 with raspi3b+ ?

QiwenLi0729, 2022-06-28:
BCM2837 chip have 1G memory , and I know it start with 0x0, but I don't know how what is the  address distribution . 0x3f000000 is peripherals base address.
How can I get the Address distribution of 1 GB memory for BCM2837 with raspi3b+ ?  bcm2837 has some difference with bcm2835. It is some troublesome to know detailed information.

lurch, 2022-06-28:
You might find https://github.com/raspberrypi/linux/blob/rpi-5.15.y/arch/arm/boot/dts/bcm2837-rpi-3-b-plus.dts a useful starting point?

QiwenLi0729, 2022-06-29:
> You might find https://github.com/raspberrypi/linux/blob/rpi-5.15.y/arch/arm/boot/dts/bcm2837-rpi-3-b-plus.dts a useful starting point?

```
// SPDX-License-Identifier: GPL-2.0
/dts-v1/;
#include "bcm2837.dtsi"
#include "bcm2836-rpi.dtsi"
#include "bcm283x-rpi-lan7515.dtsi"
#include "bcm283x-rpi-usb-host.dtsi"

/ {
	compatible = "raspberrypi,3-model-b-plus", "brcm,bcm2837";
	model = "Raspberry Pi 3 Model B+";

	chosen {
		/* 8250 auxiliary UART instead of pl011 */
		stdout-path = "serial1:115200n8";
	};

	memory@0 {
		device_type = "memory";
		reg = <0 0x40000000>;
	};

```
Thanks, This is my understanding that the memory address is start with 0x0 to 0x40000000, But the 0x3f000000 is peripherals base address.  So, the memory address space be broken ?

lurch, 2022-06-29:
ping @pelwell , who knows much more about all this than I do.

pelwell, 2022-06-29:
Although the 3B+ has a full 1GB RAM, the ARM cannot see the top 16MB of RAM because that is where the peripherals are mapped. However, this memory isn't wasted because it is occupied and used by the firmware.

pelwell, 2022-06-29:
By the way, that table is wrong - it should really say "0x3F00_0000 .. 0x3FFF_FFFF".

QiwenLi0729, 2022-06-29:
Many thanks, I got it. For BCM2837, the memory address is 0x0 - 0x3effffff.  So, the memory is actually 1008MB rather than 1024MB.

JamesH65, 2022-09-04:
Ping. Not sure if the table has been updated.

lurch, 2022-09-04:
@JamesH65 It seems exceedingly unlikely that we'll be updating https://datasheets.raspberrypi.com/bcm2836/bcm2836-peripherals.pdf ?

@QiwenLi0729 Has your question been answered, and if so can this issue be closed?

aallan, 2022-09-04:
> @JamesH65 It seems exceedingly unlikely that we'll be updating https://datasheets.raspberrypi.com/bcm2836/bcm2836-peripherals.pdf ?

We will *not* be updating this document.

closed 2022-09-04 by aallan