raspberrypi/documentation #2571: [libcamera-still]: How to split the raw data up into its red, green, and blue components?

thomaskleiven, 2022-07-02:
Is it possible to split the raw data up into its red, green, and blue components for images obtained with `libcamera-still`? Ideally, there would be a with a similar approach as described for the old `picamera` module ([docs](https://picamera.readthedocs.io/en/release-1.13/recipes2.html#raw-bayer-data-captures)).

1) For example, after capturing a raw image using the command: `libcamera-still -r -o test.jpg` - how can I extract the red, blue and green components from the `test.dng` file (before any demosaicing)?

2) Which bytes does the bayer data occupy in the `test.dng` using the 8-megapixel Camera Module v2?

3) Where in the byte stream does the header start / end?

Thanks in advance for any pointers 🥇

lurch, 2022-07-02:
ping @davidplowman

davidplowman, 2022-07-04:
You would need a software application or library that allows you to read DNG files. Obviously there are a number of raw editor and converter applications, but I don't know if any would support this feature. There are also a number of Python libraries that would allow you to read DNG files and access the data.

Alternatively, maybe you could consider using Picamera2 directly? Something like this might be what you want:
```
from picamera2 import Picamera2
picam2 = Picamera2()
picam2.configure(picam2.still_configuration(raw={}))
picam2.start()
raw = picam2.capture_array("raw")
```
Now `raw` contains the pixel array, and you can find out about the image with
```
print(picam2.camera_configuration["raw"])
print(raw.shape)
```
This will tell you the image size, the Bayer order and so on.

thomaskleiven, 2022-07-20:
Thanks for the answer! Using `picamera2` like you suggest makes sense. Also, I found that `rawpy` ([docs](https://github.com/letmaik/rawpy)) allows you to extract the red, green and blue component from images captured with the `libcamera-still -r -o test.jpg` command.

closed 2022-07-20 by thomaskleiven