raspberrypi/documentation #2575: Schematic change issue

moonplay85, 2022-07-04:
I have a problem with the changed schematic.

The Raspberry Pi 4B product was initially certified as a model before the KC certification was changed in Korea.
The problem was that I found out that the Raspberry Pi 4B had some schematic changes.

If you look at the first picture below, the picture above is a KC certified product.
That is, the model before the original schematic was changed.
However, if you look at the second picture, the red box is the changed circuit part.

This is the first photo schematic for the original 2GB and 4GB models.
The 8GB model is the second photo schematic below.
[image]

By the way, I don't know if there was a change this year, but I found out that the 4GB model was also changed to the same circuit diagram as the 8GB model.

In conclusion, there are two problems.

1. If you change the circuit diagram in Korea, you must get KC certification again.
2. If the changed circuit diagram is currently sold in Korea, it becomes a non-KC-certified product and is illegally sold.

Does anyone have any data or knowledge regarding the schematic change?
I can't check the raspberry pi documentation, so I'll leave a comment on the post.

aallan, 2022-07-04:
Product change notices and compliance certification for all Raspberry Pi models can be found in [PIP](https://pip.raspberrypi.com/).

closed 2022-07-04 by aallan

JamesH65, 2022-07-11:
We are aware that devices need to be recertified on a PCB revision change, and this is something we do as a matter of course.

moonplay85, 2022-07-12:
> We are aware that devices need to be recertified on a PCB revision change, and this is something we do as a matter of course.

So, is the certification in progress in Korea? We have a product with a changed schematic and we want to combine it with a separate product to get certification, but we need a test report with a changed schematic.

JamesH65, 2022-07-12:
> So, is the certification in progress in Korea? 

Yes.

moonplay85, 2022-07-12:
> > So, is the certification in progress in Korea?
> 
> Yes.
Thank you for your answer. In addition, I just received an NDA email again, so please approve it. 
It says that the ID is in Korean and cannot be recognized automatically. 
I have to do it manually, so please approve.