raspberrypi/documentation #2584: Give examples of reputable SD card suppliers

vedantsinghania05, 2022-07-11:
Documentation page in question: https://www.raspberrypi.com/documentation/computers/getting-started.html#troubleshooting

On the first bullet point in the troubleshooting, hyperlinking the words "reputable supplier" to a list of a few reputable suppliers could be helpful. This could be helpful for people that are confused about what SD card to purchase for their Raspberry Pi.

lurch, 2022-07-11:
We obviously couldn't hope to list **all** "reputable suppliers", and any suppliers that we don't list will complain that we don't mention them.

vedantsinghania05, 2022-07-12:
Thank you for the feedback. I will keep it in mind for future documentation changes :)