raspberrypi/documentation #2585: Toolchain mistakes python comment marker as an Asciidoc header

aallan, 2022-07-11:
The toolchain throws a `raise Exception("{} contains a Markdown-style header (i.e. '#' rather than '=')".format(filename))` error when I use Python comment characters `#` inside `[source]` blocks. Right now the toolchain is stopping us from adding comments to Python code in the documentation.

@lurch this isn't an urgent one, and slots in below your current to do list, adding directly to the back log.

lurch, 2022-07-11:
Can you work-around that by doing an `include::` for the Python file, rather than typing it directly into the AsciiDoc? :shrug:

aallan, 2022-07-11:
> Can you work-around that by doing an `include::` for the Python file, rather than typing it directly into the AsciiDoc? 🤷

Yes, almost certainly!