raspberrypi/documentation #2596: Connecting to the Internet with Raspberry Pi Pico W - use of first person plural vs. second person

mrichardson23, 2022-07-20:
There are a few spots in "Connecting to the Internet with Raspberry Pi Pico W" where first person "we" is used instead of second person "you." (The latter I believe is generally preferred for tutorials.)

For example, in section 3.7 (The MAC address) there's a note which says "We have to set the wireless active (which loads the firmware) before we can get the MAC address." To align with the style of most of the rest of the document, I would suggest "You must set the wireless active (which loads the firmware) before you can get the MAC address."

Section 3.9.2 (Controlling an LED via a web server) walks users through the process in first person plural, but section 3.10. (Running iperf) starts with "You can install iperf using the upip tool:"

aallan, 2022-07-20:
Sorry Matt, but it's intentional. I'm not a great fan of using "you" in tutorial text, and mostly default to "we" or even "I". That said, that particular book is somewhat text-lite as it was written in a bit of a rush before launch. It needs looking at.

closed 2022-07-20 by aallan

mrichardson23, 2022-07-20:
No prob! I would suggest a consistent approach, then!