raspberrypi/documentation #2605: Video Rotation

woofwoof, 2022-08-13:
Hi,

I couldn't get the --hflip option working to flip a video horizontally (using libcamera-vid streaming over rtsp to VLC) but --rotation 180 does work.

I don't think it's me!

Thanks,

WW.

lurch, 2022-08-13:
ping @davidplowman

davidplowman, 2022-08-13:
Hi, thanks for the question. To help us know where to look, could you please report:

* What kind of Raspberry Pi you are using
* The exact kind of camera that you have (v1, v2, HQ cam, or something else; also whether it's a genuine Raspberry Pi model)
* The values reported by `libcamera-hello --version`, `uname -a`, `vcgencmd version`
* The exact command line that you are using
* The console output when this command starts.

Thanks!

woofwoof, 2022-08-28:
Hello,

Sincerest apologies for the delay. I took a break! :)

It's an original Raspberry Pi Zer0 2w. It **was** a version 1 camera but since it is not working at all now so I just ordered an (original) version 2. I am running Bullseye which is up-to-date.

Libcamera-hello is telling me there's no camera now (see above)!

The command line that worked was:

libcamera-vid --rotation 180 -t 0 -inline -o - ¦ cvlc stream:///dev/stdin -sout '#trp{sdp=rtsp://[IP ADDRESS]:[PORT]/etc...

I vaguely remember prior to this I was trying the above command but with:

--hf

instead of

--rotation 180?

(Though it was a while ago now, sorry).

As I stated now the camera is not being recognised at all so I'm starting all over with the more modern one!

Hope it works and thanks for your help. Apologies again for my tardiness.

davidplowman, 2022-10-21:
Sorry for not replying, I guess I did get or didn't see any notification back at the end of August. Is there still a problem here to look at? Thanks!

woofwoof, 2022-10-30:
So sorry, I moved onto other aspects of the projects. It mainly worked for me with a new camera though I still didn't use -hflip I believe?

woofwoof, 2022-10-30:
The workaround definitely worked well. As to the original issue I cannot vouch for that but again the workaround suffices.

closed 2022-10-30 by woofwoof