raspberrypi/documentation #2607: Proposal to rename `overclocking options` section

andrum99, 2022-08-18:
Recently a user on the forums (https://forums.raspberrypi.com/viewtopic.php?p=2030369) was advised to use one of the 'overclocking' options to change the `temp_limit` parameter. This parameter is listed in the `overclocking options` section of the documentation. This is something that has come up a few times over the years, where users discover, or are pointed to, overclocking parameters which can also be used for other purposes, such as underclocking to save power, or tuning the clocks to limit the thermal output of the SoC.

I created this issue as I thought it might be worth discussing changing the title of the `overclocking options` section to something more generic. Do, of course, feel free to close this if you think this is a bad idea.

Perhaps a title such as 'adjusting system clocks` or `adjusting system clocks - overclocking and underclocking` might be more appropriate?

aallan, 2022-08-18:
This is one of those "how many angels on the head of a pin" sort of thing that I honestly don't think makes much of a difference until the overall narrative arc of the documentation is fixed. Which isn't going to be any time soon. 

I have no objections to the section titled `Overclocking options` now being called `Adjusting the system clock`. 🤷‍♂️

andrum99, 2022-08-18:
> This is one of those "how many angels on the head of a pin" sort of thing that I honestly don't think makes much of a difference until the overall narrative arc of the documentation is fixed. Which [...]

`System clock` singular is potentially problematic, since there is more than one clock. `Adjusting system clocks` also happens to be one less word, so a bit simpler.

lurch, 2022-08-18:
Would we want the text in [raspi-config](https://www.raspberrypi.com/documentation/computers/configuration.html#raspi-config) to also be updated to match? :shrug:

andrum99, 2022-08-19:
I'm going to close this, on the basis that I disagree with aallan over the wording, and as he pointed out it probably doesn't make much difference at this stage anyway.

closed 2022-08-19 by andrum99