raspberrypi/documentation #2615: CM3/CM3+ USB host boot mode enabled prevents USB device mode

giampiero7, 2022-09-07:
Hi!
Here:
https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#raspberry-pi-2b-3a-3b-cm3-cm3-zero-2-w

there is this note about the RPi 3A+:

> On the Raspberry Pi 3A+, setting the OTP bit to enable USB host boot mode will permanently prevent that Raspberry Pi from booting in USB device mode.

After enabling the OTP bit on a CM3 and a CM3+ board, it looks like the above applies in these cases too.

Is this correct? Is there any way to re-enable USB device mode on the CM3/3+?

Thanks!

lurch, 2022-09-07:
ping @ghollingworth

ghollingworth, 2022-09-07:
On Pi 3A+ the OTG_ID is permanently wired to ground (not through a pull-down)

On CM3/3+ in the CM3IO board the OTG_ID pin is brought out to a header (although there is a resistor pulling it to ground)

When the device boots, it checks the two OTP bits (one enabling USB host and the other USB device), if they are both set then it uses the OTGID pin to decide (zero would be host, 1 would be device)

So it'll never work with Pi 3A+, but it can work with the CM3 depending on how it is wired on the motherboard (you can desolder the resistor on CM3IO)

Gordon

closed 2022-09-07 by ghollingworth