raspberrypi/documentation #2687: outdated guide for `Setting up a Routed Wireless Access Point`

beta-tester, 2022-11-05:
i have the actual Raspi OS `2022-09-22-raspios-bullseye-armhf.img` version installed to my RPi Zero W and tried to set it up as an AP+STA.
for the AP part i watched the guide for [Setting up a Routed Wireless Access Point](https://www.raspberrypi.com/documentation/computers/configuration.html#setting-up-a-routed-wireless-access-point).
but it looks this guide is outdated.

there it is written `Buster` is the newest version instead of `Bullseye`.

and they use `iptables` but in the current Raspi OS `iptables` is not installed anymore.
`sudo: iptables: command not found`
and its transformation from `iptables` to `nftables` from the earlier version of RaspiOS Bullseye seems to be gone as well.

aallan, 2022-11-05:
Honestly somewhat tempted to remove this section, and the following section as well. Feels like something that can be better deal with with the new[ tutorials section](https://www.raspberrypi.com/tutorials/) of the site.

[image]

See also #2314.

lurch, 2022-11-06:
Also tangentially touches on #2520 (maybe?)

aallan, 2022-11-07:
Right. I think we should yank this, and the bridge mode section, and rewrite as a tutorial for the main site. Assigning it to @tobybroberts.

aallan, 2022-11-07:
Primarily using `NetworkManager` 'cause it's the new hotness and that's what people will have (ping @tobybroberts), but potentially both I guess depending on how much overhead of extra effort (and awkwardness) it adds to the tutorials.

aallan, 2022-11-07:
@tobybroberts wrt our conversation earlier today, we should merge the routed and bridged tutorials together into one "setting up a wireless access point".

tobybroberts, 2022-11-07:
All noted.

beta-tester, 2022-11-07:
`NetworkManager` is the new standard then i am fine when there is only a tutorial for that available. thank you in advanced,

when i can add some wishes, please also try to make a tutorial for AP + STA mode.
e.g. for the RPi Zero W it would be nice to have STA and AP simultanuosly on its wlan.
there are some old uncomplete guides out there, but none is using `NetworkManager` to do that yet.

beta-tester, 2023-01-08:
> This issue has been automatically marked as stale because it has not had recent activity. It will be closed if no further activity occurs. Thank you for your contributions.

but the tutorial section hasn't changed and is still outdated

aallan, 2023-01-08:
It's actually more or less at the top of the to do list, someone should be looking at it next week.

beta-tester, 2023-03-15:
still in progress?

aallan, 2023-04-27:
Now very much part of #2314.

aallan, 2023-07-30:
Still a thing.

aallan, 2023-08-22:
@lurch Talking to @jehlers42 right now and this part of the docs (highlighted in red) still seem to be present on the `bookworm` branch. Shouldn't all this stuff be deleted from the branch at this point. The tutorials still need to be written but I thought you'd told me you'd deleted the text from the branch at this point? If not everything in the red box needs to go. 

@jehlers42 can you figure out what text needs to go, and get rid of it from the `bookworm` branch. Post the commit back to this thread, and leave the issue open, 'cause we still need to write the updated tutorial(s) which will end up in Wordpress.

aallan, 2023-08-22:
...and they are no longer in the branch. Leaving this issue open because someone (me, @tobybroberts, Nate perhaps when he joins) needs to write replacement tutorials (in Wordpress).

lurch, 2023-08-22:
> I thought you'd told me you'd deleted the text from the branch at this point?

Nope, I don't ever remember claiming that. I was leaving the text in-situ in case it'd be a useful reference for whoever ends up writing the replacement tutorials? But you're right, they should definitely be removed from the `bookworm` branch before it gets merged :+1: 

On a vaguely-related note, I wonder if it'd be useful to have a "link-to-tutorial boxout" (like @nelliemckesson has done for PIP and the RPi Press books) that we could use to link from the documentation to the tutorials? :thinking:

aallan, 2023-08-22:
> On a vaguely-related note, I wonder if it'd be useful to have a "link-to-tutorial boxout" (like @nelliemckesson has done for PIP and the RPi Press books) that we could use to link from the documentation to the tutorials? 🤔

That is a *fine* idea. I will spin up an issue.

aallan, 2023-09-04:
@nathan-contino So this is us yanking tutorial content that shouldn't live here and writing a new one in Wordpress (and then we can use that tutorial macro stuff, see #3082 to link back to it the documentation in the right place)

aallan, 2023-10-06:
@nathan-contino Going to reopen this issue as there is still a Wordpress tutorial associated with it to write!

aallan, 2023-10-11:
Live at https://www.raspberrypi.com/documentation/computers/configuration.html#host-a-wireless-network-on-your-raspberry-pi.

There will also be a tutorial going into https://www.raspberrypi.com/tutorials/ at some point soon.

closed 2023-12-15 by aallan

beta-tester, 2023-12-16:
is the RPi onboard WiFi able to do AP + STA at the same time? and if so, how to do this to get AP + STA running?

aallan, 2023-12-16:
> is the RPi onboard WiFi able to do AP + STA at the same time? 

No.