raspberrypi/documentation #2699: Document HDMI-CEC feature disparity between full and fake KMS

cihlarma, 2022-11-20:
With the Bullseye update, full KMS is now the default graphics driver as opposed to fake KMS in Buster.

From `config.txt` in `2022-09-22-raspios-bullseye-armhf.img`:
```
# Enable DRM VC4 V3D driver
dtoverlay=vc4-kms-v3d
max_framebuffers=2
```
From `config.txt` in `2022-09-22-raspios-buster-armhf.img`:
```
[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
dtoverlay=vc4-fkms-v3d
max_framebuffers=2
```

The full KMS driver seems to not have the same level of support for HDMI-CEC as the fake KMS which automatically turns the TV on and changes to the appropriate input source without any other configuration, something I discovered this the hard way after I upgraded to Bullseye. Nowhere in the [official documentation](https://www.raspberrypi.com/documentation/) was I able to find a single mention of this, what saved me was a user post on the [forums](https://forums.raspberrypi.com/viewtopic.php?t=338709):
>This function is not supported when using the now default kms video driver. dtoverlay=vc4-kms-v3d
>If you want that function as before then you need to change to the fake kms driver.
>In config.txt change dtoverlay=vc4-kms-v3d to
>`dtoverlay=vc4-fkms-v3d`

Please consider documenting this lack of automatic HDMI-CEC functionality in full KMS somewhere at least until this feature disparity is resolved, assuming such resolution is even technically possible; announcing this in a news item like you have done with the [Widevine issue with the 64-bit OS version](https://www.raspberrypi.com/news/raspberry-pi-os-64-bit/) would be good.

JamesH65, 2022-11-20:
Rather than changing the docs, it really should be fixed, so have you reported this as an error on the kernel repo?

cihlarma, 2022-11-20:
By 'kernel repo' I assume you mean the '[linux](https://github.com/raspberrypi/linux)' repository as opposed to the '[firmware](https://github.com/raspberrypi/firmware)' one? I have not reported it in either repo; since a regular forum user not denoted as 'engineer' with a badge as RPi engineers are in the forums was already aware of the issue, I assumed this was already known internally and was on the engineering's team to-do list as part of bringing the FOSS driver to feature parity with the proprietary firmware as they have been doing over the years. I don't pay much attention to RPi development, I had simply assumed this was general knowledge that had been passed around on forums and was forgotten to be added to official docs due to its ubiquity.

JamesH65, 2022-11-20:
There is no guarantee that a forum post will make it to the issues repo. If its not an issue in the repo, then it could go missing. And yes, the Linux repo is the one.

aallan, 2022-11-20:
Also related #2310.

closed 2023-01-29 by github-actions[bot]