raspberrypi/documentation #2704: News for the News

seamusdemora, 2022-11-29:
Sorry to leave this here; I've tried to reach Simon Long about some instructions he left in [this News article](https://www.raspberrypi.com/news/the-latest-update-to-raspberry-pi-os/), but it seems my messages aren't getting through for some reason. The piece about NetworkManager in his latest post indicates that NetworkManager can be installed on any RPi via `sudo apt install network-manager` (following the obligatory apt update & full-upgrade).

I wanted to try this on my RPi 3B+ Bullseye (Lite distro) system, so I tried the install. My system locked up during the installation. I could not establish a connection following a power cycle. Fortunately I have a good backup, so I'll restore my system tomorrow.

I'm posting this to share the experience, and suggest that someone pass this along to Simon Long.

aallan, 2022-11-29:
Thanks. We're investigating internally. We'll let you know what we figure out.

spl237, 2022-11-29:
We've tested this internally, and have been unable to repeat the problem. It's very unlikely that anything in an apt install would be able to lock up a system - it is more likely that something unrelated elsewhere on your system died at the same time.

closed 2022-11-29 by lurch

seamusdemora, 2022-11-29:
@spl237 :  Did you test it on a "Lite" distro? May I ask the release date of the distro you tested? 

I agree it's unusual! I've never seen anything like this. The progress bar just froze at 79%, and didn't budge for nearly an hour. I couldn't recover control, so I pulled power. It seemed to re-boot, but I was unable to connect or get a ping response. I restored from backup earlier today, and the RPi 3B+ seems to be back to normal. 

The SD card is a fairly fresh one, and I had not seen any prior odd behavior suggesting that anything was out of order.

Thanks for the feedback.