raspberrypi/documentation #2711: Pico pinout diagram doesn't show UART1 TX & RX for GP20 & GP21

todbot, 2022-12-12:
According to the RP2040 Datasheet, there are two missing notations on the Pico pinout diagram:

- GPIO 20 - can be UART1 TX in addition to I2C0 SDA
- GPIO 21 - can be UART1 RX in addition to I2C0 SCL

[image]
[image]

aallan, 2022-12-12:
Yup. But the RTS and CTS pins for that particular pairing *aren't* available. The pin out diagram only shows UART RX/TX pins pairs where you also have RTS and CTS pins available, despite not actually showing them because most people don't care about them. 

TL;DR It's a deliberate choice I made for the diagram.

closed 2022-12-12 by aallan

todbot, 2022-12-12:
Just as a datapoint though, in case you change your mind: For a few of my MIDI-based projects, I'd struggled getting UART MIDI added because of the lack of UARTs on the right side of the Pico. It was only after checking the RP2040 datasheet did I discover. 

So for MIDI-based projects, that do not need RTS/CTS, it would've been appreciated.

lurch, 2022-12-13:
If you want to see **all** the UART pin-labels, you can click on the "Advanced" option on https://pico.pinout.xyz/