raspberrypi/documentation #2716: [TUTORIAL] Connecting via SSH/ retrieving one's IP address

epoch1970, 2022-12-20:
Some tutorials advocate (rightly, IMHO) using mDNS to connect without prior knowledge of the IP address.

However this one at least https://www.raspberrypi.com/tutorials/nas-box-raspberry-pi-tutorial/ directs users to lookup the IP address in their router.

aallan, 2022-12-20:
Thanks. That's one of the older ones. That needs fixing. Ping @tobybroberts.

lurch, 2022-12-20:
@tobybroberts You'll also need to advise the user to set a unique hostname in the Raspberry Pi Imager settings.

tobybroberts, 2023-02-19:
Still to do.

aallan, 2023-07-02:
Still a thing!

aallan, 2023-09-04:
Still needs doing.

aallan, 2023-09-16:
@tobybroberts If you have time could you duck into the tutorial and resolve this one? Shouldn't take that long I think?

tobybroberts, 2023-09-19:
Tutorial updated. Although it's worth noting that whilst the setup can use hostname.local the openmediavault webserver can only be opened using the LAN IP address.

closed 2023-09-19 by tobybroberts