raspberrypi/documentation #2717: BCM2711: AUX_MU_LCR_REG Data size field is wrong

alloncm, 2022-12-24:
The latest BCM2711 [peripherals docs](https://datasheets.raspberrypi.com/bcm2711/bcm2711-peripherals.pd) is wrong about the data size parameter.

According to this [errata for BCM2835](https://elinux.org/BCM2835_datasheet_errata#p14), bit 1 also must be set for 8 bit mode (and not just bit 0 as the docs says).
I tested this on my RPi4 and also checked the [Linux kernel headers](https://git.kernel.org/pub/scm/linux/kernel/git/next/linux-next.git/tree/include/uapi/linux/serial_reg.h#n110) and saw that this parameter is 2 bits wide and contains also options for Data size of 5 and 6 (which the docs never mention).

Thanks!

lurch, 2022-12-24:
ping @pelwell for confirmation?

pelwell, 2022-12-24:
Yup.

Tavish108, 2023-03-07:
please assign me this

pelwell, 2023-03-07:
If you wish to contribute, fork the repo, make the change to your copy, and submit a Pull Request.

lurch, 2023-03-07:
The source-code for our PDFs (including the BCM2711 Peripherals datasheet) isn't publicly available.

pelwell, 2023-03-07:
Post some alternate wording, then. Anything to help overcome the logjam.