raspberrypi/documentation #2719: USB Ports documentation

cmangla, 2022-12-27:
See documentation/asciidoc/computers/raspberry-pi/usb-bus-on-raspberry-pi.adoc

For pi4, the document clearly states how many USB ports of which version are available. For pi{zero,1,2,3}, under the paragraph "Raspberry Pi Zero, 1, 2 and 3", the same information is not explicitly noted. It would be useful to note here, for each Pi, the USB ports and versions, and possibly the connected controllers.

lurch, 2022-12-28:
Sounds similar to #2657 ?
For the benefit of anybody else reading this: [here](https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#raspberry-pi-4) is where the above-mentioned page appears in our online documentation.

aallan, 2023-05-07:
Stil valid.

aallan, 2023-07-09:
Still valid.

PR #3052 (2023-08-15): update port info for older models
Closes #2719
closed 2023-08-15 by aallan