raspberrypi/documentation #2720: legacy camera works?

demovoidgan, 2022-12-31:
Following this:
https://www.raspberrypi.com/documentation/computers/camera_software.html#re-enabling-the-legacy-stack
this is true?
I am using the latest version of Rpi IOS 64
Even if I enable the legacy camera, then I can't run the raspistill command because it doesn't exist.
Is the documentation correct?

aallan, 2023-01-01:
As far as I know the legacy camera stack is only available in the 32-bit version of the OS. Ping @davidplowman.

davidplowman, 2023-01-03:
Yes, that's correct. The legacy stack is not supported with a 64-bit OS.

aallan, 2023-01-03:
Added a warning to that effect, see #7a5d76b.

closed 2023-01-03 by aallan

lurch, 2023-01-03:
@spl237 Should `raspi-config` also check for / warn about this?

lurch, 2023-01-03:
> Added a warning to that effect, see [#7a5d76b](https://github.com/raspberrypi/documentation/commit/7a5d76ba5c186e577fa43caf7e10335fe59cb3a3).

So we now have both
```
IMPORTANT: The legacy camera stuck is not available in the 64-bit version of Raspberry Pi OS.
```
at the top of that section, and
```
WARNING: The legacy camera stack cannot be re-enabled on a 64-bit Raspberry Pi OS release.
```
at the bottom of that section? :thinking: :laughing:

spl237, 2023-01-03:
I could put a check in raspi-config to see if it is being run on 64-bit, and not offer the option on there, but it might cause more confusion than it alleviates when people can't even find the (documented) option to turn it on...

JamesH65, 2023-01-03:
There is a "stuck" instead of "stack" in the PR.

aallan, 2023-01-10:
Fixed. See [af3d4bf](https://github.com/raspberrypi/documentation/commit/af3d4bf0a72032cfdea53f6f90dff2c78a1aac70)