raspberrypi/documentation #2725: Information on composite video output is outdated

JamesH65, 2023-01-08:
There have been some changes in this area, and the docs don't actually cover composite output very well anyway, it's rather spread out over multiple places, so this needs updating. Some is factually incorrect.

Originally reported by @popcornmix.

aallan, 2023-01-08:
No modern Raspberry Pi has composite output in any case, we should probably just yank what's there.

lurch, 2023-01-08:
Does this also link to #2310 ?

JamesH65, 2023-01-08:
> ~No modern Raspberry Pi has composite output in any case, we should probably just yank what's there.~ Huh, they do. I honestly wasn't aware of that. 🤷‍♂️

The composite was moved to the 3.5mm jack on the Pi2. The Pi400 does not have the jack, so is the only one that doesn't support composite.

popcornmix, 2023-01-09:
See: https://forums.raspberrypi.com/viewtopic.php?p=2067531#p2067531

The main thing omitted currently is `,composite` (or `,composite=1`) is missing here:
```dtoverlay=vc4-kms-v3d,composite```

and you won't get any composite output without it. `raspi-config`'s enable composite option does do that for you.

Also note that config.txt settings like `sdtv_mode`/`sdtv_aspect`/`sdtv_disable_colourburst` apply only to the legacy (no-kms and fkms) display driver, not the default kms one.

andrum99, 2023-02-22:
I'm considering having a look at this one. Could I possibly ask James if he could specify the parts which he knows to be factually incorrect? I can then have a play with my Pis and figure out what needs to be said. Also following on from what James said above: if I was to write an overview of composite output on the Pi, in order to have it all in one place rather than spread over several pages, where do folk think it should be placed? And more importantly, does Alasdair think this is required?

aallan, 2023-05-07:
Still valid.

lurch, 2023-05-07:
I guess this will probably get cleaned up as part of the Bookworm changes?

aallan, 2023-07-09:
Still valid, and tagged as a Bookworm issue.

lurch, 2023-07-27:
Also needs a tweak to https://www.raspberrypi.com/documentation/computers/configuration.html#composite-video as this raspi-config option is now used to manually-enable composite video output (from KMS) on **all** models of Raspberry Pi.

lurch, 2023-08-01:
Fixed on the `bookworm` branch.

commit 400f921e (2023-08-01): Update composite-video configuration instructions.
Fixes #2725
commit acdfc4cb (2023-08-17): Update composite-video configuration instructions.
Fixes #2725
commit 0cfd1682 (2023-09-06): Update composite-video configuration instructions.
Fixes #2725
commit d1d31c81 (2023-09-07): Update composite-video configuration instructions.
Fixes #2725
closed 2023-08-01 by lurch