raspberrypi/documentation #2730: Update UF2 files for Pico W

aallan, 2023-01-10:
Need to update the UF2 files offered in the getting started sections of the documentation for Pico W, e.g. [blinking an LED](https://www.raspberrypi.com/documentation/microcontrollers/c_sdk.html#blink-an-led).

lurch, 2023-01-10:
It _does_ already say "Download the Blink UF2 [for Raspberry Pi Pico](https://datasheets.raspberrypi.com/soft/blink.uf2), or [for Pico W](https://datasheets.raspberrypi.com/soft/blink_picow.uf2)." ? :thinking:

aallan, 2023-01-10:
Yeah, I fixed that already. There are some other things that need doing.

aallan, 2023-03-02:
All fixed. I think. Maybe have forgotten one. But I think they're all fixed up now.

closed 2023-03-02 by aallan

lurch, 2023-03-02:
Just in case it's useful...

```
$ git grep "\.uf2" "*.adoc"
documentation/asciidoc/microcontrollers/c_sdk/quick_start.adoc:You now have `hello_world.elf` to load via a debugger, or `hello_world.uf2` that can be installed and run on your Raspberry Pi Pico via drag and drop.
documentation/asciidoc/microcontrollers/c_sdk/your_first_binary.adoc:. Download the Blink UF2 https://datasheets.raspberrypi.com/soft/blink.uf2[for Raspberry Pi Pico], or https://datasheets.raspberrypi.com/soft/blink_picow.uf2[for Pico W].
documentation/asciidoc/microcontrollers/c_sdk/your_first_binary.adoc:. Download the https://datasheets.raspberrypi.com/soft/hello_world.uf2['Hello World' UF2].
documentation/asciidoc/microcontrollers/debug-probe/updating-firmware.adoc:NOTE: There is currently no newer version of the firmware. The firmware running on your Debug Probe is the latest available. If you have accidentally overwritten the firmware on your Debug Probe, the https://github.com/raspberrypi/picoprobe/releases/latest/download/debugprobe.uf2[latest release of the firmware] can be found https://github.com/raspberrypi/picoprobe/releases/latest[on Github].
documentation/asciidoc/microcontrollers/micropython/drag-and-drop.adoc:* https://micropython.org/download/rp2-pico/rp2-pico-latest.uf2[Raspberry Pi Pico] 
documentation/asciidoc/microcontrollers/micropython/drag-and-drop.adoc:* https://micropython.org/download/rp2-pico-w/rp2-pico-w-latest.uf2[Raspberry Pi Pico W] (with https://makeblock-micropython-api.readthedocs.io/en/latest/public_library/Third-party-libraries/urequests.html[urequests] and https://docs.micropython.org/en/latest/reference/packages.html[upip] preinstalled)
documentation/asciidoc/microcontrollers/raspberry-pi-pico/utilities.adoc:* Download the https://github.com/raspberrypi/picoprobe/releases/latest/download/picoprobe.uf2[UF2 file]
documentation/asciidoc/microcontrollers/raspberry-pi-pico/utilities.adoc:* Download the https://datasheets.raspberrypi.com/soft/flash_nuke.uf2[UF2 file]
```