raspberrypi/documentation #2742: libcamera build instructions may have some holes?

njhollinghurst, 2023-01-18:
### Building libcamera
 - The list of dependencies to be installed before building libcamera doesn't mention `cmake`. It seems to be needed.
 - The recommended configuration (for meson) doesn't mention the `-Dpycamera` option, needed for picamera2 integration.

### libepoxy vs. libcamera-apps
 - It's mentioned near the top of **libcamera and libcamera-apps packages** that `libepoxy-dev` is required "only ... if you want support for the X11/GLES preview window". Since most users _will_ want this, could it be better emphasized, perhaps with a reminder in the **Building libcamera-apps** section?
 - Is the `libegl1-mesa-dev` dependency also required even when libepoxy is _not_ rebuilt from scratch?

aallan, 2023-01-18:
ping @davidplowman

davidplowman, 2023-01-18:
Thanks, Nick.

Is `cmake` really needed for libcamera? I'm sure it was not the case some time ago, but maybe things have changed...

Yes, we should list `-Dpycamera` (and perhaps add that you can omit it if you don't want Python bindings).

libepoxy-dev - yes, maybe we could emphasise that the other way round.

libegl1-mesa-dev... no idea. Possibly. Anyone fancy trying it??!

njhollinghurst, 2023-01-18:
Before documenting `-Dpycamera` I ought first to figure out the proper way to get its installation onto Python's path (picamera2 currently fails to load it).

aallan, 2023-05-07:
Still valid.

njhollinghurst, 2023-05-08:
Yeah I could not bring myself to document adding the build tree (rather than an installation point) to PYTHONPATH.
I might have a go at seeing if the local installation can be made to install it to a sensible place, as the `apt` version does.

lurch, 2023-05-09:
FYI the `-Dpycamera` part has been done in #2921

aallan, 2023-05-17:
Fixed in #2946

closed 2023-05-17 by aallan