raspberrypi/documentation #2746: Supports generating corresponding header files to the compiled system during manual compilation, requiring a command

BrokenWingsIcarus, 2023-01-22:
The kernel I compiled from git has no header files. I hope the community can help me and tell me how to generate header files from the compiled source files.

aallan, 2023-01-22:
Not a documentation issue. This is the sort of question you should be asking on the [forums](https://www.raspberrypi.org/forums/), so I’m closing this issue. But please follow up with further comments if you have specific questions about the [documentation](https://www.raspberrypi.org/documentation/).

BrokenWingsIcarus, 2023-01-22:
In the kernel header section, it is described that the header file has been provided in the source code library. If the kernel is compiled manually, the corresponding header file should also be copied to pi 4 at the same time, so as to obtain the complete binary kernel and its header file.

https://www.raspberrypi.com/documentation/computers/linux_kernel.html#kernel-headers

BrokenWingsIcarus, 2023-01-22:
Forgive me, I'm not a native speaker..

closed 2023-01-23 by BrokenWingsIcarus