raspberrypi/documentation #2765: building kernel question

laenzlinger, 2023-01-29:
I was trying to build a kernel and was asking had issues following the instructions in https://github.com/raspberrypi/documentation/blob/develop/documentation/asciidoc/computers/linux_kernel/building.adoc

its mentioned to call (for example for 64 bit systems)

```
KERNEL=kernel8
```

and then later
```
sudo cp arch/arm64/boot/Image.gz /boot/$KERNEL.img
```

This will line would copy the Image.gz file to /boot/.img because there is no env variable `KERNEL`  set anymore when the Image.gz file is copied.

wouldn't it be better to execute 
```
export KERNEL=kernel8
```

There must be a misunderstanding on my side here, because I guess this manual is used very often?

Feel free to correct me.

lurch, 2023-01-30:
I've not tried following those exact instructions myself (I just stick to the distro-supplied kernels), but I wondered if maybe the `sudo` command was "hiding" the `KERNEL` env-var. So on my regular Ubuntu laptop I just did:
```bash
$ KERNEL=kernel8
$ echo $KERNEL.img
kernel8.img
$ sudo echo $KERNEL.img
kernel8.img
```
and it looked okay? :confused: 

ping @pelwell as I believe he does lots of kernel-compilation...

P.S. Is it possible that you typo-ed e.g. `$KERNLE` or something? :wink:

aallan, 2023-01-30:
You only need to `export` if you're spawning a new shell, e.g.

````
$ KERNEL=foobar
$ echo $KERNEL
foobar
$ sh
$ echo $KERNEL

$ exit
````

as opposed to,

````
$ export KERNEL2=fizzboz
$ sh
$ echo $KERNEL2
fizzboz
$ exit
````

if you're sudo'ing, that's not spawning a new shell, so `$KERNEL` will be available.

````
$ KERNEL3=foobar
$ echo $KERNEL3
foobar
$ sudo echo $KERNEL3
Password:
foobar
$
````

pelwell, 2023-01-30:
What you've misunderstood is that it is the shell that replaces `$KERNEL` with `kernel8` based on its current environment, i.e.
```
$ sudo cp arch/arm64/boot/Image.gz /boot/$KERNEL.img
```
is transformed to:
```
$ sudo cp arch/arm64/boot/Image.gz /boot/kernel8.img
```
and then `sudo` is invoked to run the copy.

If you were calling another script that was accessing `$KERNEL` then you would need to export it, but that isn't the case.

Just to be clear, there is no requirement to set `KERNEL` to any value - it is just a way for these instructions to avoid some duplication when describing the build steps for multiple kernels.

closed 2023-01-30 by aallan

pelwell, 2023-01-30:
Be careful with that "spawning a subshell" distinction - this is fine:
```
$ KERNEL=foobar
$ echo $KERNEL
foobar
$ sh echo $KERNEL
foobar
```
What matters is which shell performs the expansion/substitution.

aallan, 2023-01-30:
Yup!

````
$ sh echo $KERNEL
````

is not the same as

````
$ sh
$ echo $KERNEL
````

laenzlinger, 2023-01-30:
Thank you very much for all your explanations and comments!

Today, I can not reproduce the issue anymore. It must have been a stupid typo as described by @lurch, or maybe I had another terminal open at that time, which I used by mistake.

Sorry that have wasted your time!