raspberrypi/documentation #2798: Misleading gdb command in debug probe swd docs?

veryalien, 2023-02-26:
I got caught out by a nasty 'gotcha' documentation problem, at least for my setup.

I think there is a misleading gdb command in the debug probe swd documentation.
On this page:
[https://github.com/raspberrypi/documentation/blob/develop/documentation/asciidoc/microcontrollers/debug-probe/swd-connection.adoc](https://github.com/raspberrypi/documentation/blob/develop/documentation/asciidoc/microcontrollers/debug-probe/swd-connection.adoc)
there is a sequence of commands:

Then open a second terminal window, switch to the directory containing your built binary, and start GDB.

```
$ gdb blink.elf
> target remote localhost:3333
> monitor reset init
> continue
```
I'm running a self-built from source openocd (to get a 0.11.0 version) and a downloaded 'gdb' on ubuntu.

If I do this with vanilla gdb I get a warning about unknown architecture "arm". The target then doesn't support monitor.

You really mean gdb-multiarch, don't you? It works fine with commands like gdb-multiarch blink.elf

The instructions in Chapter 6 of the pico sdk specifically mention starting gdb-multiarch and not gdb.

Maybe there's supposed to be an alias for gdb and gdb-multiarch, but on my system they appear to be different programs.

aallan, 2023-02-26:
See https://www.raspberrypi.com/documentation/microcontrollers/debug-probe.html#installing-gdb, the instructions explicitly say to install gdb-multiarch.

closed 2023-02-26 by aallan

veryalien, 2023-02-26:
I think this might need clarification.
I removed the package gdb-multiarch - which also removes the package gdb.
I did sudo apt install gdb-multiarch - which installs both gdb-multiarch and gdb packages.
If I explicitly use the command **gdb-multiarch blink.elf** then it works with the debug probe target remote localhost:3333.
If I use the command **gdb blink.elf** (as written in the debug probe swd documentation, my original complaint) then it doesn't work properly with target remote localhost:3333. It complains about the architecture.
As far as I can see, using the command gdb does not run the multiarch version.

aallan, 2023-02-26:
Fixed in #5cb848a