raspberrypi/documentation #2799: Documentation for LED 2/3/4/5 for Picoprobe

resterampeberlin, 2023-02-26:
Picoprobe contains several status LED (LED 2/3/4/5) attached to GPIOs (7/8/15/16) of the internal RP2040. However, their meaning is not documented yet.

aallan, 2023-02-26:
Do you mean the LEDs on the [Debug Probe](https://www.raspberrypi.com/documentation/microcontrollers/debug-probe.html) hardware, or for the [Pico Probe](https://github.com/raspberrypi/picoprobe) software you run on a Raspberry Pi Pico?

resterampeberlin, 2023-02-26:
I mean D2/3/4/5 in the actual probe hardware and as depicted in the schematic. I received the probe hardware yesterday, and after the first tests, I have some ideas about what those LEDs mean, but formal documentation would be great.

P33M, 2023-02-27:
D2 (green) - GPIO7 - UART_RX activity
D3 (yellow) - GPIO8 - UART_TX activity
D4 (green)  - GPIO15 - DAP_CONNECTED
D5 (yellow) - GPIO16 - DAP_TARGET_RUNNING

Note that OpenOCD just switches both DAP LEDs on when the target is connected, and turns them off when it calls DAP_DISCONNECT. Other debug hosts do more fanciful things.

aallan, 2023-02-27:
I'll add that to the documentation page.

closed 2023-02-27 by aallan