raspberrypi/documentation #2805: Doxygen: Adjust link handling

nelliemckesson, 2023-02-27:
> The big holdup with bumping pico-sdk from 1.4.0 to 1.5.0 eventually turned out to be this line (or similar!!) in the raw-doxygen HTML (in build-pico-sdk-docs/docs/doxygen/html/group__pico__cyw43__driver.html )
> <p>Initializes the lower level cyw43_driver and integrates it with the provided <a class="el" href="structasync__context.html" title="Base structure type of all async_contexts. For details about its use, see pico_async_context.">async_context</a>. </p>
> 
> I can't remember the exact details now, but IIRC it was trying to rewrite links to structasync__context.html#whatever to networking.html#something and then failing because async_context wasn't included on that page (because it's only a struct, and not a class/module). So I came up with the (slightly hacky) solution in https://github.com/raspberrypi/documentation/pull/2743/commits/6a7e141a46e58f864bbff93a2f69339d8589d25b which seems to work okay, but I dunno if there might be a "more robust" way of fixing this?  If you comment out that line in your local version of transform_doxygen_html.py you should be able to see what the problem actually is (now that the branch has been updated to Pico SDK 1.5.0)

This is getting caught in some faulty link handling, it seems. The whole way that links are currently handled needs to be re-written, imo.

nelliemckesson, 2023-03-07:
More input, from Graham:

> Clicking between “top level” items in the navigator is clunky as you switch pages; this may be hard to fix in general, but at a minimum I think we should fix the loss of context; i.e. if I go from the thing above, expand Hardware APIs, and pick a particular sub item, then the page switches, and I see vestiges of where I thought I was going with a # in the URL, but the navigation to that section does not happen

lurch, 2023-03-08:
> if I go from the thing above, expand Hardware APIs, and pick a particular sub item, then the page switches, and I see vestiges of where I thought I was going with a # in the URL, but the navigation to that section does not happen

I _think_ this is the same thing as what I described in #2829 ?

closed 2023-05-28 by github-actions[bot]