raspberrypi/documentation #2808: Doxygen: Fix formatting for section opener

nelliemckesson, 2023-02-27:
This section of the HTML-doxygen (see https://raspberrypi.github.io/pico-sdk-doxygen/group__pico__printf.html )

[image]

renders like this on the new version:

[image]

nelliemckesson, 2023-02-27:
@lurch is the problem that there's no space above the first line of normal text?

lurch, 2023-02-27:
Nope, the problem is the way the email address is getting mangled.

nelliemckesson, 2023-02-27:
Ohhhhhh @lurch my eyes were not prepared to see that, haha

PR #2847 (2023-03-08): Hide email obfuscator spans
The doxygen source appears to use some email obfuscator spans. I decided to leave them in the content but hide them via CSS. Fixes #2808
closed 2023-03-08 by aallan