raspberrypi/documentation #2810: Doxygen: Use correct top-level title

nelliemckesson, 2023-02-27:
I've just noticed that every single page under /pico-sdk/ has "Pico C SDK" as its top-level header.

[image]

Perhaps it'd be more appropriate to put the "page title" there (if we have one), so "High Level APIs" in the screenshot above?

aallan, 2023-02-27:
Yup the page title should be there, so "High Level APIs" in that particular case

nelliemckesson, 2023-02-28:
@lurch turns out this is being caused by the [asciidoc variables that are added in create_build_adoc.py](https://github.com/raspberrypi/documentation/blob/develop/scripts/create_build_adoc.py#L89-L98). The page titles are pulled from those variables, and since those variables are added before the index.json gets generated (which is how the pico-sdk subpages are included in index.json), they're just using the default page title for everything. I attempted to fix but ran into some walls related to the sequence that things are being run -- figured it might be best just to toss it over to you and let you take a stab.

PR #2830 (2023-03-06): Fix doxygen titles
Fixes #2810 
ping @nelliemckesson to check that things all look okay from her PoV.
commit 153abfda (2023-03-06): Use correct page-titles on Doxygen-created files. Fixes #2810
closed 2023-03-06 by aallan